donovan6000/M33-Fio #1: Print test border fails

phelpsw, 2015-08-13:
Using M3D Fio 0.9 and using Control -> Print test border, I receive the following error message:

```
2015-08-13 12:05:38,022 - octoprint - ERROR - Exception on /api/plugin/m3dfio [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/util/flask.py", line 414, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/api/__init__.py", line 94, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/usr/local/lib/python2.7/dist-packages/octoprint_m3dfio/__init__.py", line 367, in on_api_command
    self.getPrintInformation(temp)
  File "/usr/local/lib/python2.7/dist-packages/octoprint_m3dfio/__init__.py", line 1339, in getPrintInformation
    if self.minZExtruder < self.bedLowMinZ or self.maxZExtruder > self.bedHighMaxZ or self.maxXExtruderLow > self.bedLowMaxX or self.maxXExtruderMedium > self.bedMediumMaxX or self.maxXExtruderHigh > self.bedHighMaxX or self.maxYExtruderLow > self.bedLowMaxY or self.maxYExtruderMedium > self.bedMediumMaxY or self.maxYExtruderHigh > self.bedHighMaxY or self.minXExtruderLow < self.bedLowMinX or self.minXExtruderMedium < self.bedMediumMinX or self.minXExtruderHigh < self.bedHighMinX or self.minYExtruderLow < self.bedLowMinY or self.minYExtruderMedium < self.bedMediumMinY or self.minYExtruderHigh < self.bedHighMinY :
AttributeError: 'M3DFioPlugin' object has no attribute 'minZExtruder'
```

donovan6000, 2015-08-13:
Thanks for letting me know. It should be fixed now.

ra100, 2015-11-24:
Last time I checked, printing test border worked fine. Maybe you can close this issue.

closed 2015-11-25 by donovan6000