donovan6000/M33-Fio #4: Printer stopped in the middle of the print

santilland, 2015-09-30:
Hello donovan,
i started a little bigger print job (20.3MB) with the M3D Fio (0.15) installed as plugin on octoprint (via manual pip install). Everything was working as expected and i went away letting it print when i came back it had just stopped middle print (at 6.1MB printed) with this in the log file. 
Do you know what could be the issue? Should i adress this somewhere else?

Thanks for the awsome work!

Recv: ok 139302
Recv: ok 139303
Send: N139306 G1 X31.963608 Y47.860776 Z13.474178 E4241.907177*4
Send: N139307 G1 X33.377822 Y49.274990 Z13.490765 E4241.965382*5
Recv: ok 139304
Serial timeout while writing to serial port, trying again.
Unexpected error while writing to serial port: AttributeError: ''NoneType' object has no attribute 'write'' @ comm.py:_doSendWithoutChecksum:1639
Closing down send loop
Send: N139308 G1 X34.792035 Y50.689203 Z13.507351 E4242.023587*4
Unexpected error while writing to serial port: TypeError: ''NoneType' object is not callable' @ comm.py:_doSendWithoutChecksum:1635
Changing monitoring state from 'Printing' to 'Error: TypeError: ''NoneType' object is not callable' @ comm.py:_doSendWithoutChecksum:1635'
Recv: ok 139305
Connection closed, closing down monitor
Closing down send loop
Unexpected error while setting baudrate: 230400 SerialTimeoutException: 'Write timeout' @ comm.py:_monitor:1048
Connection closed, closing down monitor

Nold360, 2015-09-30:
Hi, 
it souldn't be related to the size, since i've already printed ~70MB gcodes...
Maybe a firmware issue? Which firmware are you using? (Printer tells it after connecting)

santilland, 2015-09-30:
Hi Nold,

im at work right now so i cant check, but i have installed the (i think latest) M3D software (2015-08-11-setup_m3dbeta-V1.3.6.3.exe) which did a firmware update. Can't find which firmware it does update to though in the readme.

donovan6000, 2015-09-30:
I've just pushed a version that increased the read and write timeouts for the serial communication. I don't think this problem was caused by M3D Fio, but maybe increasing the timeouts will prevent it from happening again.

santilland, 2015-09-30:
Thanks!
When i get back i'll update and start the print again. I'll report the results as soon as i can.

santilland, 2015-10-01:
Hi,
i let it print over night (with the new timeout setting) and everything went fine, it did the complete job. From the number of tests its not possible to say if this was the solution or even if it was a continuous problem, so if it does not hurt i would keep the change.
Thank you donovan for the quick response.

closed 2015-10-01 by santilland