donovan6000/M33-Fio #6: Feature Request: please provide gcode without bed compensation for the gcode viewer but with for printing

mrvn, 2015-10-09:
Loading bed compensated gcode view in octoprint is completly useless because it things every z correction is it's own layer. And switching off bed correction for preview and on again for print is a lot of extra work and easy to forget to turn it back on. The plugin should save the gcode without bed correction for the viewer and with for printing.

dragondgold, 2015-10-11:
The built-in G-Code Viewer renders the selected file to print, so I don't think there is a way to view it correctly. A workaround could be to save the save both g-codes, the original and the processed one with some prefix like "processed_<filename>" so you could view the original g-code and print the processed one. 

Or you can just view the model in your computer and then send it to octoprint. I use Slic3r and it has a built-in octoprint support so you can send the g-code to your printer in just one click.

mrvn, 2015-10-19:
Actually there is, you just have to use different hooks.

What I mean is that there is a hook that gets called when gcode gets send to the printer and you can modify the gcode then. So bed leveling should not be done during slicing but during printing on the fly. The same would apply to backlash compensation.

donovan6000, 2015-12-02:
Implemented an on the fly G-code pre-processing settings in the latest release, V0.22, which allows OctoPrint's G-code viewer to work as expected.

dragondgold, 2015-12-02:
This should be closed now!

closed 2015-12-03 by donovan6000