donovan6000/M33-Fio #8: print speed

bjoernbarnekow, 2015-11-10:
Hi Donovan,

I got some stange problem and want to ask if anyone can reproduce this or if this is only my problem.

I got a M3D micro, a raspberry pi 2, octoprint 1.2.7 and m3d fio 0.17. 

A small testcube with some details on it prints usually in 1 hour, with octopi 7 hours. Then I figured out what’s going wrong. Maybe this helps to fix this bug

The gcode
'G1 X10’ moves the printhead 10mm right. works fine.
‘G1 X10 F3000’ should move it 10mm right with 3000mm/min or 50mm/sec.
‘G1 X10 F300’ should do the same with 300mm/min or 5mm/sec.

It turnes out that F300 moves the printhead ways faster than F3000. So I think the Fxxxx will be interpreted as duration fot the step instead of the speed.

Can you reproduce this?

Thanks a lot.

regards,
Björn

donovan6000, 2015-11-10:
The printer's speed is kinda weird, and it's due to the printer's firmware. The feedrates are converted during the G-code preprocessing in order to compensate the speeds to match that the printer expects (I just copied what the official software does for it). Here's the algorithm:

ConvertedValue = 30 + (1 - (CurrentValue / 60) / 60.0001) * 800

So F300 becomes F763, and F3000 becomes F163. So your results make sense since F300 moves way faster than F3000 after the conversion.

I tend to have slow prints too. I'm not entirely sure why OctoPrint + M3D Fio is so much slower than the official software though since they both do this same conversion. The Cura profiles that M3D Fio provides are based off of the profiles generated by the official software, so the speeds shouldn't differ by that much. It might be a result of differences in the Cura versions (the official software uses the old V15.02.1 Cura).

bjoernbarnekow, 2015-11-11:
Hi Donovan,

in Cura I can set the print speed in mm/sec, Im my case I choose 20mm/sec for printing, 5mm/sec for the first layer and 40mm/sec for movements without printing. This should be F1200, F300 and F2400. The resulting gcode look like this:

G0 F2400
...
G1 F300 X54.200 Y52.542 E0.00431
...
G1 F600 X55.003 Y51.450 E3.01737

The normal print speed is the half of my value. The rest looks ok to me. Or is cura doing wrong?

OMG. I just tried the M3D console and F300 is ways slower than F3000. In microprint it is the other way.

http://reprap.org/wiki/G-code#G0_.26_G1:_Move tells me F3000 should be faster. What if M3D is wrong? And you use their code. What would happen if you disable the conversion?

thanks.
Björn

donovan6000, 2015-11-11:
Cura's G-code works as expected on nearly all 3D printers, so Cura isn't doing anything wrong here.

From what I understand Microprint uses on the fly conversion on all G-codes, so it's impossible to run G-code using Microprint without it being converted. However both the official software and M3D Fio don't have that annoyance, so any commands you test with them will give you the actual result. That is why running F300 in the M3D console will produce what the printer's firmware considers F300 to be, and running F300 in Microprint will run it at what the printer's firmware considers F763 to be since it gets converted before being sent to the printer.

I would consider M3D to be wrong, but they've never said that they're using reprap's G-code. So I guess their proprietary G-code is supposed to have bizarre F values lol. In this post, gearsawe attempted to deduce what the F values actually mean for the Micro M3D. It's pretty impressive since it was before we knew the algorithm used by the official software.
https://printm3d.com/portal/forum/index#/discussion/1929/f-value-in-m3d-gcode-what-does-it-mean

bjoernbarnekow, 2015-11-11:
Oh my goodness! I want the same drugs the M3D developers used to take when writing this.

Ok. I will try following: I make the gcode with cura the same way and convert the F values manually. The printing driver or driver subsystem converts it back to my original values. Then I test it again.

One more thing: When starting a new print job with octopi it first moves the print head to the beck right corner for calibration and moves down to the bed. This process takes more than 40 minutes. Maybe this is a result of the speed settings in the gcode file.

Thanks a lot.

donovan6000, 2015-11-11:
Just found this post by Michael Armani , the founder of M3D, that slightly explains what the F values mean. It's also kinda funny how he praises the skill level of his programming team XD. He said, "Our team is very skilled, they wrote the complete firmware from scratch and all of the GUI, a bug like this won't take long to fix once it is surfaced..." and "There may also be long-standing confusion here as to standard Gcode vs. what we use with M3D software.  Our hardware is not based on any existing firmware and was written from scratch to accommodate a very unique hardware stack.  As a result we are processing speed as delays, not as feedrate, and do not currently process backlash in the firmware like most printers do.  The numbers you see as a feedrate in post-processed gcode by our spooler are faster when the numbers are lower."
https://printm3d.com/forums/discussion/comment/16672/#Comment_16672

So maybe F300 could mean make that movement take 300ms?

And any STL you slice/upload to OctoPrint while M3D Fio is installed will go through a series of pre-processing stages. So you can disable M3D Fio's 'Feedrate Conversion' pre-processor stage under OctoPrint's settings to stop it from converting the F values. And to see what command will actually be sent to the printer when printing a pre-processed file, look for the fully processed file in ~/.octoprint/uploads/.

Not sure why that's taking 40 minutes (should take about 15 seconds). Try running the G28 command manually, which is what a starting print uses to re-center the print head, and see how long that takes. M3D Fio's 'Preparation' pre-processor stage will add in G-code to make it leave excess filament on the print head in front left corner before it starts printing. Is that contributing to the 40 minutes?

bjoernbarnekow, 2015-11-12:
Hi Donovan,

this explaines a lot. Thanks to the skilled software engineers the other axes move as expected. ;-)

I made a small perl programm to convert the F values. So I can use cura to make the code, convert it an send it to octopi. All movements work as expected now.

I also figured out why it took so long to start. The F3000 command sets the printer to something like the slowest possible movement. With the new codes these wrong F values do no longer disturb the printer. Thanks for your help.

One more thing: When I disconnect the printer, octoprint is not able to reconnect. I have to reboot the raspberry pi and hope for a connection.

Thanks for your help.

regards,
Björn

> Just found this post by Michael Armani , the founder of M3D, that slightly explains what the F values mean. It's also kinda funny how he praises the skill level of his programming team XD. He said, [...]

Jetzt aber Shell!

jamesarm97, 2015-11-12:
I have to just say how dumb this is that they took something that was halfway a standard (F value) and completely changed it. Why couldn’t they just adapt their code to use what everyone else does. Sometimes reinventing something from scratch isn’t the best way to go.

- James

bjoernbarnekow, 2015-11-12:
Hi,

I wrote me a converter.

http://eris.bytecamp.net/M3DF.pl

Now it works.

Thanks and regards,
Björn Barnekow

> 
> I have to just say how dumb this is that they took something that was halfway a standard (F value) and completely changed it. Why couldn’t they just adapt their code to use what everyone else does. Sometimes reinventing something from scratch isn’t the best way to go.
> 
> - James
> 
> >
> > Hi Donovan,
> >
> > this explaines a lot. Thanks to the skilled software engineers the other axes move as expected. ;-)
> >
> >
> >
> >
> > I made a small perl programm to convert the F values. So I can use cura to make the code, convert it an send it to octopi. All movements work as expected now.
> >
> > I also figured out why it took so long to start. The F3000 command sets the printer to something like the slowest possible movement. With the new codes these wrong F values do no longer disturb the printer. Thanks for your help.
> >
> > One more thing: When I disconnect the printer, octoprint is not able to reconnect. I have to reboot the raspberry pi and hope for a connection.
> >
> >
> > Thanks for your help.
> >
> > regards,
> > Björn
> >
> >
> > >
> > > Just found this post by Michael Armani , the founder of M3D, that slightly explains what the F values mean. It's also kinda funny how he praises the skill level of his programming team XD. He said, "Our team is very skilled, they wrote the complete firmware from scratch and all of the GUI, a bug like this won't take long to fix once it is surfaced..." and "There may also be long-standing confusion here as to standard Gcode vs. what we use with M3D software. Our hardware is not based on any existing firmware and was written from scratch to accommodate a very unique hardware stack. As a result we are processing speed as delays, not as feedrate, and do not currently process backlash in the firmware like most printers do. The numbers you see as a feedrate in post-processed gcode by our spooler are faster when the numbers are lower."
> > > https://printm3d.com/forums/discussion/comment/16672/#Comment_16672
> > >
> > > So maybe F300 could mean make that movement take 300ms?
> > >
> > > And any STL you slice/upload to OctoPrint while M3D Fio is installed will go through a series of pre-processing stages. So you can disable M3D Fio's 'Feedrate Conversion' pre-processor stage under OctoPrint's settings to stop it from converting the F values. And to see what command will actually be sent to the printer when printing a pre-processed file, look for the fully processed file in ~/.octoprint/uploads/.
> > >
> > > Not sure why that's taking 40 minutes (should take about 15 seconds). Try running the G28 command manually, which is what a starting print uses to re-center the print head, and see how long that takes. M3D Fio's 'Preparation' pre-processor stage will add in G-code to make it leave excess filament on the print head in front left corner before it starts printing. Is that contributing to the 40 minutes?
> > >
> > >
> >
> > Jetzt aber Shell!

Jetzt aber Shell!

donovan6000, 2015-11-12:
@bjoernbarnekow Awesome perl script! I noticed the problem with disconnecting the printer too. There's currently a bug in OctoPrint where it doesn't fire the 'octoprint.events.Events.DISCONNECTED' event when the printer is disconnected, and M3D Fio relies on that event to be able to reconnect after disconnecting. I'm not sure when this issue started since I was using OctoPrint V1.2.4 for a long time before updating a few days ago.

@jamesarm97 I feel like the only reason they do some of this non-standard stuff to reduce the size of their printer's firmware. Their latest firmware size is 31906 bytes, and the max flash size of the ATxmega32C4 is 32768 bytes. So they don't have much wiggle room anymore.

bjoernbarnekow, 2015-11-13:
Hi Donovan,

I could imagine the number stored in the F value is cpu cycles per microstep. This would make sense. This would keep the firmware easy. The ATmega is an interesting cpu because it can solve all commands in exactly one cycle. Counting the assembler commands can be used for timing.

The perl script is completely useless. I found the ‘feed rate conversion’ switch in the fio plugin settings. This makes exactly the same. So the hint for everyone who asks why the printer is so slow - enable the 'feed rate conversion’ and upload the gcode files again.

M3D tells us the max speed is 80mm/sec. With the feed rate conversion formula makes moves faster than 62mm/sec impossible. If you set X to 56mm/sec and Y to 56mm/sec and both axes move with max speed the resulting speed is 80mm/sec. They measure the max speed diagonally.

Rebooting the raspberry pi does not solve the reconnect. The printer crashes after ‘disconnecting’ from octoprint. Have to repower and recalibrate before use again. Strange.

> @bjoernbarnekow Awesome perl script! I noticed the problem with disconnecting the printer too. There's currently a bug in OctoPrint where it doesn't fire the 'octoprint.events.Events.DISCONNECTED' [...]

regards,
Björn Barnekow

foosel, 2015-11-27:
> There's currently a bug in OctoPrint where it doesn't fire the 'octoprint.events.Events.DISCONNECTED' event when the printer is disconnected

And where's my bug report for that? ;)

dragondgold, 2015-11-27:
@foosel I'm also getting the issue sometimes, will try to reproduce it and get the logs. From and old log I have this is the only thing that looks bad:

```
2015-11-27 14:09:48,590 - tornado.general - WARNING - Write error on <socket._socketobject object at 0xaee45a08>: [Errno 113] No route to host
2015-11-27 14:09:49,724 - tornado.general - WARNING - Write error on <socket._socketobject object at 0xaee863b0>: [Errno 113] No route to host
2015-11-27 14:09:49,870 - tornado.general - WARNING - error on read
```

And the terminal is just stuck in:

```
Changing monitoring state from 'Connecting' to 'Closed'
Connection closed, closing down monitor
```

donovan6000, 2015-11-27:
@foosel I thought it was bug, but it turns out I just redefined the event handler function twice and didn't realize it XD. I wasn't going to submit a bug report without being certain that the problem really existed. And thanks for helping me get the register_callback handler working a few months ago :)

@dragondgold That issue should be fixed with M3D Fio V0.19. The printer will occasionally crash when switching between bootlaoder and G-code processing mode, which will also cause this issue. But that's entirely the printers firmware's fault.

closed 2015-12-02 by donovan6000

bjoernbarnekow, 2015-12-07:
Hi,

thanks a lot. Something changed. My raspberry pi2 is not fast enough to make the calculations in realtime. Turning the new switch off the bed compensation is done during upload. But now it takes much longer than before. For a 5 MB Gcode I have to wait more than half an hour. What can this be?

Thanks and regards,
Björn

> 
> Closed #8.

Jetzt aber Shell!

donovan6000, 2015-12-07:
After implementing the on the fly pre-processor, I got rid of the shared library pre-processor since I didn't think anyone would be using it any more and it was difficult to maintain. So it will take longer to pre-process the entire file than it did before since it's using the Python functions now instead of the shared library functions. I can bring back the shared library pre-processor if it's necessary.

What issues is your Raspberry Pi having with the on the fly pre-processor?

bjoernbarnekow, 2015-12-08:
Hi,

the on the fly preprocessor takes more calculating power than available. So the print stops every few seconds. The surface gets a lot of pimples.

Left:preprocessed, right:on the fly

you can watch it during print: http://212.204.60.6/m3d.mp4

thanks.

regards,
Björn

> After implementing the on the fly pre-processor, I got rid of the shared library pre-processor since I didn't think anyone would be using it any more and it was difficult to maintain. So it will [...]

Jetzt aber Shell!

donovan6000, 2015-12-08:
Ok, I brought back the shared library pre-processor. It also gets used when pre-processing on the fly, so hopefully that should make the on the fly pre-processing run better on the Raspberry Pi.

bjoernbarnekow, 2015-12-08:
Wow,

thanks a lot. Now printing with on the fly processing and 12% cpu load. Works fine.

Merry Christmas,
Björn

donovan6000, 2015-12-09:
Awesome! Glad it's working now.

bjoernbarnekow, 2015-12-09:
Hi,

it’s me again. The problem ist still there. The cpu load is down below 15%. But it’s still not fast enough for continously printing. The priner can print for 20 seconds followed by a 6 second break, then 20 sec print, 6 sec break and so on. It seems that the on the fly calculation render a buffer and send this to the printer. Is there a way to render the next buffer in background in advande and switch the buffers when the first on is empty?

thanks a lot and best regards,
Björn