donovan6000/M33-Fio #9: Printer stops extruding shortly after start

npag, 2015-11-13:
Hi!
My issue is that after a while of printing, the printer stops extruding filament but the printhead keeps moving and rises in height for every layer.
If I restart the print, it will stop extruding relatively precise where it stopped the last time. Every model has an other point where it stops extruding.
The lastest M3D software is working fine though.
Do you have any tips on which information I should provide that could be helpful?

Your plugin is very cool, I love it already!

Nold360, 2015-11-14:
Hi,

i've the same issue. I think it could be related to "too much retraction", but I really don't know.. 
Some models print fine, some never get finished.. but it doesn't stop exactly after the same time for me.. 

Last model that failed: https://www.thingiverse.com/thing:497464

ra100, 2015-11-24:
I think, I've got that same problem. I thought the problem was with overheating. But when I forgot to check the printer and after few layers got "printed" without extrusion I found out, that filament was unloaded. Maybe this gets fixed in 2015-11-18-setup_m3dbeta-V1.3.6.6?

dragondgold, 2015-11-24:
Are you printing with PLA or ABS? This happened to me and the problem was overheating. The heat from the heater goes up and the filament gets soft, so the extruder motor gets stuck. The solution is to add a little fan to cool down the extruder, you can use this as a fan holder, http://www.thingiverse.com/thing:1106544

ra100, 2015-11-24:
I'm using PLA, and I tried to print at 200 or 194°C, and result was the same as with 230°C (I know, it's little too high for PLA). I think I haven't had problem with extrusion before, I managed to print whole 3DBenchy model with v0.16 (I think), but I got 2 horizotal shifts. And with new M3D-Fio (or better new profiles?) I can't get past ~4 mm of height. I also tried to add fan to the front of the head. Not the fancy way you have, first I'd have to somehow print that cover :)
I'm not saying it can't be HW problem. Just trying every option.

ep1cman, 2015-11-24:
I can second this, I can't get any prints above a mm or two. It then starts printing "bobbly" until it stops together. It seems to be retracting the filament too much but then not extruding enough because if you manual extrude after stopping the print it takes quite a while for filament to reach the nozzle.

donovan6000, 2015-11-25:
Disabling retraction will probably fix this issue. You can edit the profiles that M3D Fio provided which are located in ~/.octoprint/slicingProfiles/cura/

I was hoping that people would have experiment and improved the profiles by now, but I've yet to get any input...

ra100, 2015-11-25:
I'll try adjusting profile, I just wasn't sure which options are "safe" to change. I think I should play with `retraction_speed` and `retraction_amount`. I'll post results.

ep1cman, 2015-11-25:
So I have had a chance to experiment with the profile setting and not much seems to help. I have tried disabling retraction, increasing flow rates, slowing down movement. None of these seem to solve the problem. I have also tried to copy the profile from the latest m3d software without much luck. I have noticed the m3d-fio seem to ignore the temperature setting in the settings pane, it seem get reset too. Last night I put both the profile and setting to 180 but when i started printing it was working at 190. It might well be something to do with overheating as dragondgold said but I don't see why the official software does not have the same issue.

ra100, 2015-11-25:
I think, that first layer is printed at 10°C more than other layer, it should be heat bonding. And yes, temperature in slicing profile is ignored, don't know why. If you want to print on other than default temp, you have to change it in M3D-Fio settings.
And I created repository with my slicing profiles, feel free to do whatever you want :) https://github.com/ra100/M3D-slicing-profiles

donovan6000, 2015-11-25:
When thermal bonding is enabled, all temperature and fan G-codes are removed. It'll use the temperature set in the M3D Fio tab of OctoPrint's settings, and set the first layer to 10 degrees higher and the second to 5 degrees higher. This is how the official software does it. So you can disable thermal bonding and your temperatures will be used.

The official software uses Cura V15.02.1 which probably accounts for some differences.

ra100, 2015-11-25:
With me, I think @Nold360 was right. I tried to add fan to front head cover, and I was able to get through print above 2cm high. But what I noticed, that I had to turn fan on "manually". Fan started before print, during warming phase, but in the moment when extrusion started, the fan went off. I am printing with heat bonding, if that makes any difference, why fan could go off.

ep1cman, 2015-11-26:
@ra100 I tried your suggestion of manually turning on the fan during the run and now I get perfect prints. It's definitely a overheating issue because the fan I not turned on properly.

Also in regards to the print temperature,@donovan6000, like you said earlier "It'll use the temperature set in the M3D Fio tab". The issue I seem to now be having is that the temperature keeps setting itself back to 215C

donovan6000, 2015-11-28:
Cura's was adding in some M107 commands which turns the fan off right at the beginning of the print. I've reworked how the thermal bonding pre-processors removes commands, so the issue with the fan being turned off should be fixed now.

@ep1cman Is is able to save the settings for any of the other plugins? Try deleting config.yaml and users.yaml in ~/.octoprint/ and see if the temperature stops setting itself back to 215C. If it still does it, then OctoPrint isn't able to save the plugin's settings which is most likely due to permission errors.

npag, 2015-11-28:
I would like to test the fix but I don't really know hot to apply the update. Do I have to reinstall the plugin in OctoPrint? I tried checking for updates and it says there is v0.21 available but I don't see an option to install it.

donovan6000, 2015-11-28:
@npag Yeah, you can just reinstall it and it'll install the latest version of the plugin.

npag, 2015-11-28:
I am still having this issue. I am currently using v0.21 is the fix already included in this version?
I just loaded up a file and started printing. Do I need to do something extra?

donovan6000, 2015-11-28:
Yes the fix is included in V0.21. Make sure you have the preparation and thermal bonding pre-processors enabled for it to work. Can you send me the fully pre-processed G-code file that your trying to print? It's in ~/.octoprint/uploads/

npag, 2015-11-29:
Ok now I get it, I didn't reslice the model I wanted to print. Everything is working fine now, keep up the good work! Thank you

closed 2015-12-02 by donovan6000

raab6, 2016-02-28:
Have this issue with version 0.27.1.
It stops extruding but still moves to the sliced path of the file.
I have thermal bonding activated.
Any idea what I can try?
Print settings:
- 215°C (for PLA)
- Medium quality
- highest infill (which is selectable)
- always with support
- without raft
- retraction off

Thanks!

donovan6000, 2016-02-29:
@raab6 Make sure you have the preparation pre-processor enabled, the validation pre-processor's remove fan commands option enabled, and the filament type is set to PLA in M3D Fio's settings. That will set the fan's speed to max to prevent any fan commands in the G-code file from stopping the fan.

raab6, 2016-02-29:
Thanks, will check this. But could not recreate this issue. Perhaps the
timelapse caused it. Disabled it and there was no issue with extruding
anymore. Nevertheless thanks for the fast reply and good work!!! M3D Fio is
awesome!