donovan6000/M33-Fio #11: Can't print with new release

dragondgold, 2015-12-02:
Just installed your new release let me tell you it's great! We now can use the GCode Viewer and you don't have to slice every time you change something like bed leveling!

But now I can't even print, once the hotend finished heating and the printer has to move the printer just sends "Resend <line number>", M3D-Fio keeps sending the following lines and the printer stays still while Octoprint send all the GCode!

donovan6000, 2015-12-03:
That's weird! It should be able to handling resending commands without any issues. Does it do that every time you try to print? What command is the one that it needs resent?

ra100, 2015-12-03:
Seems I've got the same problem. I'm not sure which command, but it looks like command just before temperature hits desired value, I receive "T -1".

Nold360, 2015-12-04:
@ra100 - If it would be the baud rate, M3D-Fio shouldn't be communication with the printer in the first place.

ra100, 2015-12-04:
@Nold360 yep, it's working now, and I feel dumb :D But it's interesting that I was able to print on 250k baud rate before on 1.2.7, but on 1.3.0 I couldn't connect to printer.

dragondgold, 2015-12-04:
Sorry for the delay! Here is the terminal output @donovan6000 

```
Recv: T:186.2672
Recv: T:186.2672
Recv: T:186.2672
Recv: T:187.8659
Recv: T:187.8659
Recv: T:186.2672
Recv: T:-1
Recv: ok 64
Send: N73 G1 X74.985 Y47.163 E14.35619*107
Recv: Resend:65
Recv: wait
Send: N74 G1 X80.566 Y56.830 E15.08690*103
Recv: Resend:65
Recv: wait
Send: N75 G1 X81.385 Y59.081 E15.24374*96
Recv: Resend:65
Recv: wait
Send: N76 G1 X41.927000 Y44.730000 F563.333778 **108
Recv: Resend:65
Recv: wait
Send: N77 G1 X42.444000 Y44.297000 Z0.214762 E2.04413 F696.666889 **72
Recv: Resend:65
Recv: wait
Send: N78 G1 X44.654000 Y43.020000 Z0.196427 E2.21124 **56
Recv: Resend:65
Recv: wait
Send: N79 G1 X81.502 Y59.745 E15.28791*105
Recv: Resend:65
Recv: wait
Send: N80 G1 X46.533425 Y42.336068 Z0.183514 E2.342171 **4
Recv: Resend:65
```

After the printer finished heating up, the first movement is ALWAYS asking for resend, no matter what piece I try to print

donovan6000, 2015-12-05:
I'm honestly not sure what's causing that. M3D Fio lets OctoPrint handle all the line numbering and resending. Does the terminal ever show any commands with an 'N65' being sent, or does it just jump from 'N64' to 'N66'?

[This line](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L1507) is where all the commands being sent to the printer actually get sent to the printer. Hopefully that can help you debug the problem further.

dragondgold, 2015-12-07:
Yeah it's very strange, I will try a clean install of Octoprint just in case. In the terminal output I showed above, in this part:

```
Recv: ok 64
Send: N73 G1 X74.985 Y47.163 E14.35619*107
Recv: Resend:65
```

It sends the line 73 after line 64, it's missing sending the lines 65, 66, 67, 68, 69, 70, 71, 72 and 73 right?

donovan6000, 2015-12-07:
@dragondgold That's normal. The printer has an internal buffer where it can store 10 to 15 of the next G-code commands that it needs to run. This is a common things for 3D printer's to do, and OctoPrint knows about it too and will send new commands without waiting for confirmations that previous commands have finished. The printer will respond with a 'ok 64' as soon as it finished running N64, then it'll check it's internal buffer for the next command and respond with 'Resend 65' if it isn't N65. It's common for the internal buffer to become full or miss commands when the printer's running long duration operations, like waiting for the temperature to reach a certain value. However OctoPrint keeps a list of all commands that it has already sent to the printer, so it knows how to resend commands when the printer responds with a 'Resend'.

So if you look higher up in the terminal output, you should see N65, N66, N67, N68, etc somewhere.

Also, which OctoPrint version are you using?

dragondgold, 2015-12-07:
@donovan6000  Right I forgot about the buffering! I was using Octoprint 1.2.7 and a few minutes ago I did a clean install with the new 1.2.8 release but the problem persists, the strange thing is all the previous commands before heating the printer works perfect.

donovan6000, 2015-12-07:
I've been using V1.2.7, so that's definitely not the problem. What's really weird is that OctoPrint has no sufficient history of an N65 command since I've never seen OctoPrint skip line numbers. I just changed M3D Fio so that it'll write to OctoPrint's log all the commands that get sent to the printer, so try out the newest code and hopefully that helps us see where the problem is.

dragondgold, 2015-12-08:
Ok, so here is the terminal output when printing:

```
Send: N1 M107*36
Recv: ok 0
Recv: wait
Recv: ok 1
Send: N2 M104 S195*106
Send: N3 G28*16
Recv: ok 2
Send: N4 G1 Z5*99
Recv: ok 3
Recv: ok 4
Send: N5 M109 S195*96
Recv: ok 5
Send: N6 G21*28
Send: N7 G90*23
Recv: ok 6
Send: N8 M82*17
Recv: ok 7
Recv: ok 8
Send: N9 G92 E0*78
Recv: ok 9
Send: N10 G1 Z0.350 F2400.000*53
Recv: ok 10
Send: N11 G1 E-2.00000 F2400.00000*2
Recv: ok 11
Send: N12 G92 E0*116
Send: N13 G1 X41.600 Y44.983 F2400.000*100
Recv: ok 12
Send: N14 G1 E2.00000 F2400.00000*42
Recv: ok 13
Recv: ok 14
Send: N15 G1 X42.117 Y44.550 E2.04413 F600.000*25
Recv: ok 15
Send: N16 G1 X44.327 Y43.273 E2.21124*84
Send: N17 G1 X54.816 Y39.456 E2.94196*86
Recv: ok 16
Send: N18 G1 X57.330 Y39.012 E3.10906*93
Recv: ok 17
Send: N19 G1 X58.004 Y39.012 E3.15319*86
Recv: ok 18
Send: N20 G1 X60.518 Y39.456 E3.32030*86
Recv: ok 19
Send: N21 G1 X71.007 Y43.273 E4.05101 F600.000*30
Recv: ok 20
Send: N22 G1 X73.217 Y44.550 E4.21812*94
Recv: ok 21
Send: N23 G1 X73.734 Y44.983 E4.26225*80
Send: N24 G1 X75.374 Y46.938 E4.42934*90
Recv: ok 22
Recv: ok 23
Send: N25 G1 X80.955 Y56.605 E5.16005*83
Recv: ok 24
Send: N26 G1 X81.828 Y59.003 E5.32712*82
Recv: ok 25
Send: N27 G1 X81.945 Y59.667 E5.37129*86
Recv: ok 26
Send: N28 G1 X81.945 Y62.220 E5.53839*92
Recv: ok 27
Send: N29 G1 X80.007 Y73.212 E6.26911*88
Recv: ok 28
Recv: ok 29
Send: N30 G1 X79.134 Y75.610 E6.43618*82
Send: N31 G1 X78.797 Y76.194 E6.48032*80
Recv: ok 30
Recv: ok 31
Send: N32 G1 X77.156 Y78.150 E6.64744*89
Send: N33 G1 X68.606 Y85.325 E7.37815*90
Recv: ok 32
[... 213 lines cut ...]
2015-12-08 14:55:22,698 - octoprint.plugins.m3dfio - INFO - N85 G1 X50.292274 Y40.968204 Z0.241009 E2.604032
```

All the lines are being sent but it immediately fails after line 70 which sets the hotend temperature. However, now that we have the shared library back (which speeds up a LOT the collecting information before printing) it works perfect with the shared library activated! I even got a `Resend` and it's handled perfectly:

```
Recv: T:186.1366
Recv: T:186.1366
Recv: T:187.7191
Recv: T:187.7191
Recv: T:187.7191
Recv: T:-1
Recv: ok 12
Send: N15 G90 **46
Recv: Resend:13
Send: N13 G21*40
Recv: ok 13
Recv: wait
Send: N14 G90*37
Recv: ok 14
Send: N15 G90 **46
Recv: ok 15 
```

It's very strange. I don't have activated the `Preparation`, `Thermal Bonding` and `Wave Bonding` pre-processor but that shouldn't change anything.

donovan6000, 2015-12-09:
Well at least everything's working now. I guess the raspberry pi just isn't capable of meeting the resource requirements of the Python on the fly pre-processor. Glad I brought back the shared library!

closed 2015-12-09 by donovan6000