donovan6000/M33-Fio #14: Slicing doesn't work - Invalid cross-device link

ra100, 2015-12-07:
Now I'm getting this error
```
2015-12-07 17:20:25,453 - octoprint - ERROR - Exception on /plugin/m3dfio/upload [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python2.7/dist-packages/octoprint_m3dfio/__init__.py", line 4244, in upload
    os.rename(modelLocation, modelTemp)
OSError: [Errno 18] Invalid cross-device link
```

donovan6000, 2015-12-07:
I guess that can happen sometimes when using os.rename. People say using shutil.move instead will fix it, so I just updated M3D Fio to use that function instead.

ra100, 2015-12-07:
Great, it works now.

closed 2015-12-07 by ra100