donovan6000/M33-Fio #15: Slicing profile gets edleted after slicing

ra100, 2015-12-07:
I know, that tihs is not finished yet, but when I use my custom slicing profile, it gets removed after model is sliced. I need to copy that profile again to profiles folder.

donovan6000, 2015-12-07:
How the slicing works is that it'll backup the original slicer profile, model, and printer profile before replacing them so that the slicer uses the modified version of each. Then it should restored the backups after the slicer succeeds, fails, or it cancels. So I'm not sure why it's not restoring your slicer profile. I'll look into this more later today.

donovan6000, 2015-12-07:
Haven't been able to recreate this issue. Do you have a link to the profile your using?

ra100, 2015-12-07:
This is the profile https://github.com/ra100/M3D-slicing-profiles/blob/master/cura/m3d_pla_0.1mm_brim.profile, it's just modified m3d_pla_0.1mm.profile, changed temperature and platform_adhesion

ra100, 2015-12-11:
tried it again, seems OK.

closed 2015-12-11 by ra100