donovan6000/M33-Fio #16: error 'M3DFioPlugin' object has no attribute 'printerBacklashX'

ra100, 2015-12-07:
It does seem, everytihng works, but OctoPrint log/console output I see this error.
```
2015-12-07 18:52:38,296 - octoprint - ERROR - Exception on /api/plugin/m3dfio [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/media/nidhogg/pi/devel/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.9.dev1+g547dcdd-py2.7.egg/octoprint/server/util/flask.py", line 491, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/media/nidhogg/pi/devel/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.9.dev1+g547dcdd-py2.7.egg/octoprint/server/api/__init__.py", line 96, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.22-py2.7.egg/octoprint_m3dfio/__init__.py", line 897, in on_api_command
    self._printer.commands(self.getSaveCommands())
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.22-py2.7.egg/octoprint_m3dfio/__init__.py", line 2439, in getSaveCommands
    if self.printerBacklashX != softwareBacklashX :
AttributeError: 'M3DFioPlugin' object has no attribute 'printerBacklashX'

```

donovan6000, 2015-12-07:
I'm not really sure how it managed to run the getSaveCommands function without self.printerBacklashX being defined, but it should be fixed now.

ra100, 2015-12-07:
Seems fixed, great job.

closed 2015-12-07 by ra100