donovan6000/M33-Fio #17: OctoPrint disconnects on Calibartion

ra100, 2015-12-07:
I've had this problem also with older versions. Everytime I try to Calibrate Z0, bed orientation or complete calibration. Calibratino starts, but after few seconds I get message, that I'm disconnected from server. I can reconnect after a while, but also after half an hour, the calibration "modal" doesn't hide. I have to refresh window. Seem that calibration gets done, but with Complete calibration it gets stuck after first "step" (maybe center Z0?) and I don't get any instructions. I can't find anything wrong in logs.

donovan6000, 2015-12-07:
Yeah, complete calibration starts off with calibrating center Z0.

In OctoPrint settings, in the 'Serial Connection' tab under 'Long running commands' make sure it has G28, G30, G32. If it already does, then I have no idea what's causing the you to get disconnected.

ra100, 2015-12-07:
These commands are set. Than it's maybe just problem of resources. And the printer stays connected, but OctoPrint fronted "dies" or halts for 3-5 minutes. I'll try to free more resources, stop unnecessary service on RPi2.

ra100, 2015-12-08:
So I tried to stop services and let only octoprint running. No difference. I see you added logging, where can I see those logs?

donovan6000, 2015-12-08:
Logs are in ~/.octoprint/logs/octoprint.log. Right now it just logs every command that gets sent to the printer.

Does it also disconnect if you manually run a G30 command?

ra100, 2015-12-08:
When running G30 manually, everything seems to work fine.

donovan6000, 2015-12-08:
I'm still not sure why it's having issues. You can try removing commands that the calibration uses until it works so we can see exactly what's causing the problem.
https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/static/js/m3dfio.js#L1922-L1932

Occasionally some commands won't go through so I added a delay and a stop which seems to fix that, G4 P100 and M0. The M104 S0, G91, G0 Y20 Z2 F150, M109 S150, M140 S0, M107, G30 are what the official software uses when calibrating the bed. The M117 at the end is to make sure that M3D Fio knows the current Z location since it gets used by some other stuff.

In the code you'll see some weird commands. M65537 gets converted by M3D Fio to M0, and M65536 just lets M3D Fio know to wait until all the commands have been processed. I had to do some weird things at times to work around OctoPrint.

ra100, 2015-12-12:
It's definitely my Raspberry Pi problem, maybe proxy configuration. I tried it on PC, and it works fine.

donovan6000, 2015-12-15:
I just set a timeout of 60 minutes on all the ajax request, so hopefully that fixes this. Otherwise it's an issue with your network's configuration.

ra100, 2015-12-16:
I didn't have time to test it, but I'll report back, when I do.

ra100, 2015-12-16:
Thanks for the effort, problem must be somewhere in my network configuration.
As you can see in this log.
```
2015-12-16 19:36:26,168 - octoprint.plugins.m3dfio - INFO - M107
2015-12-16 19:36:26,175 - octoprint.plugins.m3dfio - INFO - G30
2015-12-16 19:40:10,521 - octoprint.plugins.m3dfio - INFO - M117
2015-12-16 19:40:10,530 - octoprint.plugins.m3dfio - INFO - M105
2015-12-16 19:40:10,618 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
2015-12-16 19:40:11,824 - octoprint.server.util.sockjs - INFO - New connection from client: 127.0.0.1
```
Connection is restored right after M105, what I think is the time, where frontend should get message, that calibration is done a then close calibration message/show it is done.

I have one other thing to try, now I have octoprint behing nginx, I'll try it directly from local network. I could be problem with nginx configuration and socket.io.
Well, if nobody else has this problem, this ticket should be maybe closed.

ra100, 2015-12-16:
Sorry for the trouble, it is problem of my proxy settings. When I tried to connec direcly //ip:port, it works fine.

closed 2015-12-16 by ra100