donovan6000/M33-Fio #18: octoprint calibration offset (z-axis)

3dfan, 2015-12-08:
Hello

My octopi + micro3d setup is almost working fine. It seems the automatic calibration is a bit off and the print is not sticking to the board.
Can I somehow manually offset the z0 in octoprint (like you can on the native micro3d software) ?

On the octoprint forum they suggested to make a custom gcode control, but they had doubts how this would integrate with the custom plugin for the Micro ...

Any suggestions welcome.

thx for helping out :)

donovan6000, 2015-12-08:
In OctoPrint's settings under the M3D Fio tab there's a setting called 'Bed Height Offset' which does what your looking for. All the 'offset' settings are equivalent to the official software's cat bed leveling screen settings.

3dfan, 2015-12-08:
coool !!!
thank you very much, I will check it tonight :D

closed 2015-12-13 by donovan6000

briankb, 2017-04-07:
is that option available for any printer profile or is it a plugin specifically for M3D? I need to adjust the z-height as well for the same reason on my prusa i3 clone. ideally a Global Z-Height offset would be available per printer profile