donovan6000/M33-Fio #20: Extruder temperature / Filament temperature / manual control

domwakeling, 2015-12-08:
Trying to troubleshoot my temperature issues ...
* am I right in thinking that the "Filament Temperature" setting in the "M3D FIO" tab of the Settings page is intended to set the general target temperature? (subject to any pre-processor or manual changes that are made)

* does the temperature setting here then pre-set the temperature control in the "Extruder" section of the Control tab?

donovan6000, 2015-12-09:
The 'Filament Temperature' setting is the only setting that M3D Fio uses. The 'Preparation' pre-processor adds in some G-code at the start of a print which sets the temperature to that value. The 'Thermal Bonding' pre-processor will remove any existing temperature related G-codes and set the temperature of the first layer to be 10 to 15 degrees hotter than that value.

The temperature control in the 'Control' tab is never set by M3D Fio nor does it update in real time. The only place in OctoPrint to see the current temperature is in the 'Temperature' tab.

domwakeling, 2015-12-09:
OK - so if the temperature isn't set in the M3D settings it defaults to 100C, and I take it that then the GCODE will default to a minimum of 150C. Explains the issue with printing cutting down to 150C, hopefully sorted.

closed 2015-12-09 by domwakeling