donovan6000/M33-Fio #23: Bed calibration not saved

ra100, 2015-12-12:
I tried to run Calibrate Bed orietation more times and the values in M3DFio settings stayed the same. Also M3D-Linux showed that same values. Thank I tried to run M107, G32, M117 commands in M3D-Linux and those values finally changed, but in OctoPrint I see the old values. Seems like "Obtain printer settings automatically when a Micro 3D printer connects" doesn't work as expected.

ra100, 2015-12-12:
Maybe related to https://github.com/donovan6000/M3D-Fio/issues/24

donovan6000, 2015-12-13:
Should be fixed now. I was just forgetting to request the the new bed orientation values after running G32.

closed 2015-12-15 by donovan6000

ra100, 2016-01-24:
I think the problem is back. After complete calibration, values are not saved. And m3d-linux can't read values:
```
M3D Linux V0.16

Attempting to connect to the printer
Connected to printer
Firmware Version: 2016010812
Serial Number: BK1504290110****
Failed to collect printer information
```

Here is serial.log http://pastebin.com/hEvmuDFj

donovan6000, 2016-01-26:
Should be fixed now. It was working on OctoPrint 1.3.0, but not on 1.2.8.

And M3D Linux isn't compatible with the newer printer firmwares since they switched from a virtual EEPROM system to a real EEPROM system and now require sending the amount of bytes you want to receive with the M618 and M619 commands. I don't really use M3D Linux for anything anymore aside from flashing [iMe](https://github.com/donovan6000/iMe), so I doubt I'll get around to updating it to work with the newer firmwares.

ra100, 2016-01-26:
Just tried it, it works. Thanks. And also m3d-linux explanation, I am calmer now, when I know, that it's nothing wrong :)