donovan6000/M33-Fio #27: on-the-fly conversion inserts more and more G4 gcodes

mrvn, 2015-12-13:
I'm printing a Dalek salt and pepper shaker and I noticed that as time goes on the printer pauses for longer and longer times mid print. Looking at the octoprint stdout I see:

2015-12-13 19:55:02,257 - octoprint.plugins.m3dfio - INFO - N369472 G4
2015-12-13 19:55:02,289 - octoprint.plugins.m3dfio - INFO - N369473 G4
2015-12-13 19:55:02,302 - octoprint.plugins.m3dfio - INFO - N369474 G4
2015-12-13 19:55:02,321 - octoprint.plugins.m3dfio - INFO - N369475 G4
2015-12-13 19:55:02,339 - octoprint.plugins.m3dfio - INFO - N369476 G4
2015-12-13 19:55:02,357 - octoprint.plugins.m3dfio - INFO - N369477 G4

I see runs of this going on and on for 400-500 lines and they seem to get longer.

mrvn, 2015-12-15:
See pull request #29

donovan6000, 2015-12-17:
You fixed this one.

closed 2015-12-17 by donovan6000