donovan6000/M33-Fio #28: Zits in the upper region using octopi and M3D-Fio

mabawsa, 2015-12-15:
Hi,
Thanks for the wonderful plugin, I am experiencing an issue though with a raspberry pi B (2015-12-15_2015-11-21-octopi-jessie-0.13.0.zip) and printing. I get lots of zits in the upper region (see attached image).
This is the test print to calibrate backlash (https://printm3d.com/solutions/article.php?id=80).
On the left is the octopi print, on the right is from M3D's software.
I have tried many settings and they all produce the same results (the print on the left uses the default plugin settings).
Steps tried:
1) Overclock raspberry pi (no change)
2) Use stable octopi 1.2 (no change)
3) Use the latest git from here (no change, like the autodetect feature though).

Am I missing something obvious?

Dougie
[image]

donovan6000, 2015-12-15:
The backlash G-code files, backlash_calibration_pla.gcode and backlash_calibration_abs.gcode, don't contain a Cura header comment and layer comments which makes it so that the official software doesn't really pre-process them like it does other files. M3D Fio doesn't rely on comments in the G-code since I didn't want to limit it's users to only using Cura, so the backlash files will be pre-processed when printing with.

So to print those files as close as possible to the official software, you need to disable all 8 of the pre-processors in OctoPrint's settings under the M3D Fio tab. This will make it so that the G-code file is printed without any modifications which is how those files were meant to be printed.

donovan6000, 2015-12-15:
Just added a button under the Calibration section called 'Print backlash calibration cylinder' that'll print a the cylinder how it was meant to be printer. That way people don't have to worry about disabling and enabling the pre-processors when doing backlash tests.

mabawsa, 2015-12-16:
Thanks, this is a great new feature, however the zits are still there. 
Blue indicates the print from the new feature:
1: Factory firmware
2: Updated firmeware
The other two are previous prints
I did measure the average time for a cycle of the cylinder for the upper and lower regions and found that an average of 10 cycles in the lower region (no zits) was 2.5s and in the upper region 4s (teenager). Is this meant to happen?
[image]

donovan6000, 2015-12-16:
Try disabling the 'Pre-process on the fly' option in OctoPrint's settings under the M3D Fio tab. There might be some issues with it, and some people have said that the raspberry pi isn't powerful enough to handle it.

The cylinder's diameter does slowly increase, which explains the longer print times for the higher regions.
[backlash](https://cloud.githubusercontent.com/assets/6620767/11838972/4365f802-a39f-11e5-889e-a804c246a76b.png)

mabawsa, 2015-12-16:
Fantastic, you hit the nail on the head. The pre-process was causing the zits. Many thanks...

donovan6000, 2015-12-17:
Thanks too!

closed 2015-12-17 by donovan6000

mabawsa, 2015-12-18:
Would the new pre-processor commit influence the raspberrypi latency?

donovan6000, 2015-12-18:
The new changes are based on [mrvn's work around](https://github.com/donovan6000/M3D-Fio/pull/29), so on the fly pre-processing should work better on the raspberry pi now. The old method would cause unnecessary delays which is what was causing the zits, however the new method doesn't produce any delays. Give it a try and let me know the results!