donovan6000/M33-Fio #30: on-the-fly commands are out of order

mrvn, 2015-12-25:
2015-12-25 17:31:48,059 - SERIAL - DEBUG - Send: N40 G1 X34.350338 Y37.338277 Z0.065165 E2.257620**40
2015-12-25 17:31:48,199 - SERIAL - DEBUG - Recv: ok 40
2015-12-25 17:31:48,201 - SERIAL - DEBUG - Send: N41 G1 X36.252507 Y36.720416 Z0.067163 E2.337615**42
2015-12-25 17:31:48,341 - SERIAL - DEBUG - Recv: ok 41
2015-12-25 17:31:48,343 - SERIAL - DEBUG - Send: N42 G1 X38.154676 Y36.102554 Z0.069160 E2.417609**35
2015-12-25 17:31:48,483 - SERIAL - DEBUG - Recv: ok 42
2015-12-25 17:31:48,484 - SERIAL - DEBUG - Send: N43 G1 X40.373000 Y35.382000 Z0.069993 E2.51090**20
2015-12-25 17:31:48,625 - SERIAL - DEBUG - Recv: ok 43
2015-12-25 17:31:48,626 - SERIAL - DEBUG - Send: N44 M105*23
2015-12-25 17:31:48,791 - SERIAL - DEBUG - Recv: ok 44
2015-12-25 17:31:48,792 - SERIAL - DEBUG - Recv: T:240.0716
2015-12-25 17:31:48,793 - SERIAL - DEBUG - Send: N45 G1 X35.295 Y31.445 E2.59128*89
2015-12-25 17:31:48,796 - SERIAL - DEBUG - Recv: ok 45
2015-12-25 17:31:48,797 - SERIAL - DEBUG - Send: N46 G1 X39.256 Y31.280 E2.74986*85
2015-12-25 17:31:48,944 - SERIAL - DEBUG - Recv: ok 46
2015-12-25 17:31:48,945 - SERIAL - DEBUG - Send: N47 G1 X67.929 Y31.280 E3.89677*94
2015-12-25 17:31:49,237 - SERIAL - DEBUG - Recv: ok 47
2015-12-25 17:31:49,238 - SERIAL - DEBUG - Send: N48 G1 X50.311000 Y34.859000 Z0.056362 E2.909859**39
2015-12-25 17:31:49,385 - SERIAL - DEBUG - Recv: ok 48
2015-12-25 17:31:49,386 - SERIAL - DEBUG - Send: N49 G1 X52.311000 Y34.859000 Z0.054643 E2.989858**41
2015-12-25 17:31:49,534 - SERIAL - DEBUG - Recv: ok 49
2015-12-25 17:31:49,535 - SERIAL - DEBUG - Send: N50 G1 X54.311000 Y34.859000 Z0.052924 E3.069857**38
2015-12-25 17:31:49,682 - SERIAL - DEBUG - Recv: ok 50
2015-12-25 17:31:49,685 - SERIAL - DEBUG - Send: N51 G1 X56.311000 Y34.859000 Z0.051205 E3.149857**45
2015-12-25 17:31:49,830 - SERIAL - DEBUG - Recv: ok 51

If you look at N47 you can see that the position is a big step from N46 and the E value is larger. But then N49.. draws an earlier line with lower E values.

mrvn, 2015-12-25:
this is with commit ff9fd586424cc77d3de41e45a2c068099da7ba22

donovan6000, 2015-12-26:
I'll look more into this after the holidays. Can you upload the Gcode file too?

donovan6000, 2016-01-10:
I think I've finally fixed this issue now. Just ran a 15 minute print and all the commands were sent in the right order. OctoPrint uses two queues for sending commands, so I had to empty both the queues before inserting new commands to guarantee that they get sent in the correct order.

closed 2016-03-15 by donovan6000

mrvn, 2016-03-16:
octoprint accepted my patch so the hook can return mutliple commands. So this sending back commands into the queue is no longer needed and the queues don't have to be emptied. Have you tried using that?

donovan6000, 2016-03-16:
Which OctoPrint branch is it implemented in?