donovan6000/M33-Fio #33: Octoprint UI not working with m3dFio settings in config.yaml

ra100, 2015-12-31:
After octoprint restart, I must first remove m3dfio settings from config.yaml to make it work again. Otherwise it doesn't who "Settings" menu and "Connection" panel. And I see nothing suspicious in logs.
Is there a way to enable m3dfio debug log?
This happens on master, maintenance and devel branch of octoprint.

ra100, 2015-12-31:
For some reason I've had offset values set to ```-nan```

donovan6000, 2016-01-01:
Not really sure how you managed to get one of the offsets to be NAN :/

There's now NAN checks for all the floating point values obtained from the printer. If the value is NAN, then it'll be set to the default values defined in the 'get_settings_defaults' function.

It also seems like this'll fix issue #34 since that was also caused by the offset being NAN, so I'll close them both if can confirm that it's fixed.

ra100, 2016-01-01:
Seems to work fine. But I can't again simulate that weird state and I have no idea how I got there. Maybe printer disconnected before saving Z values when calibrating or something. I don't know.

Either way, you can close this as solved. Thanks. And issue #34 is related, so with this, it's also fixed.

closed 2016-01-01 by donovan6000