donovan6000/M33-Fio #35: Modifications made in 3D preview are ignored

npag, 2016-01-02:
When applying changes to the model position on the printbed the printer will ignore this changes and print the model in the original place.

donovan6000, 2016-01-03:
It should be fixed now. Also make sure that the 'Center Model' pre-processor is disabled in OctoPrint's settings under the M3D Fio tab.

npag, 2016-01-05:
I'll be able to take a look later this week. Giving feedback soon!

npag, 2016-01-06:
I need to reinstall octoprint I guess. My pip installation has gone missing and I am not able to update octoprint to 1.2.8 either. I will test your fix next week.

npag, 2016-01-12:
I figured out my issues with OctoPrint and tested your modifications. Works great now, thank you very much!

closed 2016-01-12 by npag