donovan6000/M33-Fio #36: Print Quality

Fullmono, 2016-01-06:
Hi, 

I usually generate my model with Cura under windows, and send it to the printer with M3D spooler.
I'm trying M3D-Fio with the same model, and the print quality is poor. Printer's movement are not the same, M3D-Fio do a lot of random movement... 

I don't understand since it come from the same .gcode.

The question is, does Fio use M3D preprocessor when receiving a gcode from octroprint? 

Printer is also slower while printing...

donovan6000, 2016-01-10:
M3D Fio provides an on the fly pre-processor which processes each commands as it's sent to the printer. You can also disable on the fly pre-processing which will cause all Gcode files to get processed as soon as they are uploaded to OctoPrint. You can find all of M3D Fio's settings in OctoPrint's settings under the M3D Fio tab.

There was an issue with the on the fly pre-processor which could have been responsible for the random movements and slower printing. This has been fixed in the newest version of M3D Fio, V0.24.

donovan6000, 2016-03-15:
I'm probably going to close this issue in the next couple of days since no ones responded to it in a few months. Let me know if you still have any interest in testing/solving it.

closed 2016-03-18 by donovan6000