donovan6000/M33-Fio #37: Weird values in printer settings

ra100, 2016-01-08:
After update (Jan 8) I get weird values in plugin settings:
[image]
I already tried flash firmware and complete calibration.

donovan6000, 2016-01-08:
There seems to be a lot of issues with the newest M3D software, V1.4.0.6, and printer firmware, 2015122112, and no one has said anything good about it. So I reverted M3D Fio back to using a more stable printer firmware, 2015102303, until M3D releases another update.

M3D Fio is supposed to tell you that the printer's firmware is incompatible if your using anything but 2015102303, but that didn't work. It's fixed now, so it should tell you to update the printer's firmware once the printer connects and then it'll be able to read the correct values for the plugin settings.

Did you change any of the settings or use any of the calibration buttons after you saw the weird values? If so, send me a screen shot of the first 0x215 values of the EEPROM so that I can make sure nothing got messed up.

ra100, 2016-01-09:
Thanks.
Autoupdate didn't start, I've had tu flash firmware from file. Firmware button still says: "Update firmware to V2015122112". But manual flashing worked. Values are now OK, but I'm not sure about Backlash X/Y  values, I've got there 0.3 and 1.0799mm.
~~Now after connecting, I'm getting message, that I should update to newer fw version (2015122112).~~
Correction: I've had to remove `build`, `dist` and `OctoPrint_M3DFio.egg-info` folders before installing plugin. I always install plugin by `sudo python setup.py install`.
I've tried to run calibration and tried to change temperature. Here are my EEPROM values.
[image]

donovan6000, 2016-01-10:
So I decided to support the new firmware version, V2015122112, after all. I experimented with it more, and I really like how it uses standard feed rates since modifications to a Cura profile setting like bottom_layer_speed now work like they should. And OctoPrint's estimated print time is accurate now because of it.

I just released a new version of M3D Fio, V0.24, which includes the V2015122112 firmware. M3D hasn't said much about the new firmware yet, but I did a complete bed calibration anyway just in case they changed anything relating to it. Sorry for making you update your firmware again.

Your EEPROM looks fine. I just wanted to make sure that it didn't write to any of the currently unused addresses or overwrite any existing values incorrectly. Those backlash values seem plausible. The expected range for X is 0.2 to 0.6 and for Y is 0.4 to 1.3. The newest M3D Fio now includes support for M3D's [new backlash calibration procedure](https://printm3d.com/solutions/article.php?id=80]), and details about the it get displayed before printing the backlash calibration cylinder. [People have said good things about it](https://printm3d.com/portal/forum/index#/discussion/comment/27853), so it could probably help you fine tune your backlash if you need to.

ra100, 2016-01-10:
That's no problem, flashing firmware :) I'm printing right now after calibrations, so it seems everything is fine. Thanks again.

closed 2016-01-10 by ra100