donovan6000/M33-Fio #38: Printer unresponsive after 40 mins of printing

djecken, 2016-01-09:
Printer becomes unresponsive after 40 minutes of printing, although octroprint terminal still shows that it's sending and receiving commands.

ra100, 2016-01-09:
I don't think it's time related problem. I was able to print 2h30m print and it worked fine. But sometimes printer stops responding before print or after first commands, octoprint acts like everything is fine, but printer but head is not moving and I have to disconnect printer or restart octoprint.

donovan6000, 2016-01-10:
I do feel like this is more of an issue with OctoPrint than it is with M3D Fio since it's OctoPrint's responsibility to correctly handle all the sending and receiving.

This issue is difficult to recreate. Does it always happen around 40 minutes? Can you post of screenshot of OctoPrint's terminal tab when it happens?

djecken, 2016-01-10:
Looks like this is an issue with Simplify3D, sending F3600 commands M3D doesn't understand:

Recv: ok 65534
Send: N66226 G1 X45.242000 Y63.344000 F1500.000000*96
Recv: ok 65535
Send: N66227 G1 X45.112000 Y63.327000 Z24.417925 E2.1804 F3600*103
Recv: T:224.6990 U:0.5625
Recv: wait
Send: N66228 G1 X45.029000 Y63.292000 Z24.417494 E2.1836 F3600*104
Recv: T:212.3181 U:1.0000
Recv: wait

Switched to using Cura and works just fine! You can close this issue.

donovan6000, 2016-01-10:
Could you upload the Gcode file that produces this issue? It's related to the line numbers, and it's probably fixable.

M3D uses a compressed Gcode format that only supports 2 byte line numbers. So the 'Recv: ok 65535' is the highest line number value that the printer can return, and M3D Fio increments a counter to compensate for the next received response 'Recv: ok 0' so that OctoPrint sees it as 'Recv: ok 65536'. But that doesn't seem to be working in this case for some reason.

Off topic, but I just saw your [First of the New Year 3D Hangouts](https://www.youtube.com/watch?v=_nUwX_uRfLY). Aside from mispronouncing M3D Fio and my name, it was pretty good. I added images to M3D Fio's README.md like you guys suggested. The company M3D has released 3 new software versions and 2 new firmware versions for their printer since the last time that [tomasf's microprint](https://github.com/tomasf/microprint) received any major updates, so it's pretty outdated for you guys to still be recommending it. OctoPrint + M3D Fio runs perfectly fine on Windows, OS X, and Linux, so I'd encourage recommending that or the official software. Keep up the good work! :)

djecken, 2016-01-10:
Sure, sorry about butchering your name! We'll be showing off your plugin in this weeks videos as well :D

Thanks for all the hard work your putting in!
[adabot-knight-topB-m3d.zip](https://github.com/donovan6000/M3D-Fio/files/85164/adabot-knight-topB-m3d.zip)

LogicalChaos, 2016-01-10:
Running into the exact same problem, M3D never acks when the command number gets over 65535.  16 bit number anyone?  [MicroPrint](https://github.com/tomasf/microprint) rolls the command id every 100.  I noticed it with a larger print that was using bed compensation.  Turned that off, the print completed fine (at id 57k)

donovan6000, 2016-01-11:
Just released a hot fix for M3D Fio, V0.24.1, that fixes this issue.

There's some off-by-one error in the newest printer firmware that makes it think the command after N65535 is N65536 when it should think that it is N0 because it uses 16bit line numbers, and this caused M3D Fio to attempt to resend a command that didn't exist.

ra100, 2016-01-11:
Nice, this explains why, after calibrating and some printing, printer head stopped moving. :+1:

donovan6000, 2016-01-12:
So this got me looking into another issue where the printer seemed to occasionally pause for a moment while printing. Turns out this is normal behavior when the printer's newest firmware processes a G92 E0 command, but it wont send back an 'ok' for that command about 5% of the time for some reason.

I'm not sure if the printer will forget to send back 'ok's for any other commands, so M3D Fio now accounts for forgotten 'ok's since this could break M3D Fio in certain cases. So here's another hot fix, V0.24.2.

donovan6000, 2016-01-12:
Ran into another issue with resending commands who have line numbers that have rolled over. The entire communication system with resending and confirming commands should be fixed now, so ignore the V0.24.2 that I mentioned and use V0.24.3 that I just released.

LogicalChaos, 2016-01-14:
Works great for me

closed 2016-01-17 by donovan6000