donovan6000/M33-Fio #39: High memory usage on model preparation

ra100, 2016-01-10:
When trying to slice large model 3MB and more, the memory usage is over 250 MB, which can be sometimes problem on Raspberry Pi. I tried 3MB and 4MB model, and both consumed about same ammout of memory, with 1MB it has been much less. I didn't do any exact measurements, just watched htop after hittin "Slice", most memory usage was while "Applying changes" phase.
It is not as issue as suggestion for performance improvement.

[image]

donovan6000, 2016-01-12:
Should be fixed now.

I'm not really sure why the method I was using to upload the files was causing such high ram usage, but switching to the method that OctoPrint uses when uploading files seems to have fixed it.

ra100, 2016-01-12:
Works like a charm :)

closed 2016-01-12 by ra100