donovan6000/M33-Fio #40: Cannot print this gcode, error indicates print too large

LogicalChaos, 2016-01-11:
This vertical 80x80x6 lithoframe fails to print with an error of print too large.   The M3D Printer Profile in OctoPi is 113x107x112.  When I print the same gcode file in [MicroPrint](https://github.com/tomasf/microprint), it prints as expected.  The slicer doesn't appear to make a difference, tried Simplify3D and Cura (external).  I've not tried the embedded Cura engine.

[litho.gcode.zip](https://github.com/donovan6000/M3D-Fio/files/85252/litho.gcode.zip)

donovan6000, 2016-01-13:
M3D Fio enforces the same three tier boundary limitations as the official software, which has the top tier shifted toward the left for some reason. I probably should strive away from the boundaries that M3D's software uses since they're kinda weird.
[image]

microprint seems to enforce the boundary limitations from this image, which I don't think the official software has ever used as boundary limitations.
[image]

As a temporary fix, you can enable the 'Ignore print dimension limitations' option in OctoPrint settings under the M3D Fio tab which will allow you to print anything.

LogicalChaos, 2016-01-13:
> On Jan 12, 2016, at 8:27 PM, donovan6000 <notifications@github.com> wrote:
> 
> M3D Fio enforces the same three tier boundary limitations as the official software, which has the top tier shifted toward the left for some reason. I probably should strive away from the boundaries that M3D's software uses since they're kinda weird.
> 
Hmmm…  When I throw the stl into the embedded slicer, even translated over, it never fits.  The object is 80x80, so it should.  I have to rotate around Z (45 works) after translating to get it happy.  I’d not seen the bounding box before in the Modify Model pane.  Nice work!
> microprint seems to enforce the boundary limitations from this image, which I don't think the official software has ever used as boundary limitations.
> 
This is the one I’m familiar with.
> <https://camo.githubusercontent.com/614dc63ff2b9bffb174b28ef4a87b08f784d497a/68747470733a2f2f696d616765732d6e612e73736c2d696d616765732d616d617a6f6e2e636f6d2f696d616765732f532f61706c75732d73656c6c65722 [...]

I had tried that, and the print went south almost immediately.  Although that was on .23.  I’ll try it again on .24.3

Thanks,
Rob

donovan6000, 2016-01-14:
So I did some tests today, and ended up deciding on these new boundaries.
[image]

The litho.gcode files still goes out of bounds with them though because of the command 'G1 X0 F2160' on line 455977. This moves the extruder to X0 when it's in the top tier, and the lowest X can be there is 7. So if you change it to something with a higher X, like 'G1 X54 F2160', then it'll print.

LogicalChaos, 2016-01-16:
I'll try this out when I pull later today.  Thanks.

LogicalChaos, 2016-01-17:
Yup, that takes care of it.  Thanks.  I'm presuming you know about the odd little step up in the bounding box at the front?

closed 2016-01-17 by LogicalChaos

donovan6000, 2016-01-17:
That new boundary box is based on where the tip of the extruder can physically be. It can go slightly past the front of the print bed which would allow it to print on the actual frame of the printer, so the step up at the front is to make sure that the start of the print is on the bed and not the frame.

LogicalChaos, 2016-01-18:
Makes sense.  I have a nice divot on the frame of the printer showing where it can go :-)