donovan6000/M33-Fio #42: OctoPrint 1.3.0 model loading fail

ra100, 2016-01-12:
Just in case you'll be working on support for OctoPrint 1.3.0. I get error in console when loading stl file:
```
tornado.access - WARNING - 404 GET /downloads/files/local/filter 
```
It seems that somewhere in the process extension gets lost, url should be `/downloads/files/local/filter`.

From time to time I am trying to run M3DFio with `dev` branch of OctoPrint.

donovan6000, 2016-01-13:
The model editor should be working now on the latest OctoPrint dev version. There's seems to be some other issues with running M3D Fio on it though like how I'm only able to successfully connect to the printer about half of the time.

ra100, 2016-01-13:
I thought the connection problem was fixed, I've been able to connect few times in a row yesterday.

donovan6000, 2016-01-13:
Just looked into it, and it's fixed now. OctoPrint V1.3.0 sends 'N0 M110 N0' when first connecting to the printer instead of 'M110'. It also throws a bunch of serial exceptions when connecting, but those don't seem to affect OctoPrint or M3D Fio so it doesn't really matter.

So the only thing I've noticed that doesn't work in OctoPrint V1.3.0 is displaying multiple 'wait' responses in the terminal tab. They should get displayed like 'wait . . . . . .' but  each dot gets displayed on a new line. This is purely aesthetic so it also doesn't really matter that much.

ra100, 2016-01-13:
Again, awesome. I'll test i, when I get from work.
And those serial exceptions, I noticed them too, in 1.2.8, 1.2.9 and 1.3.0, but seems it doesn't affect functionality.

donovan6000, 2016-01-15:
I was still only managing to successfully connect to the printer about 70% of the time. So I started looking through the differences between OctoPrint 1.2.9 and 1.3.0, and I managed to track down what was causing the issue. OctoPrint now [waits for all commands to be sent](https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/util/comm.py#L513-L514) to the printer before fully closes the serial connection, and the Micro 3D can't respond to commands when it's in bootloader mode so it would just hang here forever.

So all my connections issues are solved now, an I'm getting 100% connection success rate with OctoPrint V1.3.0 :)

ra100, 2016-01-15:
I've been able to check it just now. And it works.

closed 2016-01-15 by ra100