donovan6000/M33-Fio #43: Tall print stops 3-4 layers from the top

LogicalChaos, 2016-01-14:
Twice in a row, with different gcode files (slight changes to the model), the print stops 3-4 layers from the top, about the same place.  The head is stuck to the print where it just stopped printing/moving.  I'm running another print logging the serial communication.  0.24.3.

[Frame_fail.gcode.zip](https://github.com/donovan6000/M3D-Fio/files/90541/Frame_fail.gcode.zip)

donovan6000, 2016-01-15:
Every time OctoPrint receives a 'wait' response, it sends another command. So it can sometimes be thousand of commands ahead of what the printer is currently processing. This also caused the G-code viewer to be further ahead that it's supposed to be, and M3D Fio would break if it ever got to being more than 65535 command ahead.

I think it was yesterday that I implemented (limiting the amount of commands that can be simultaneously sent to the printer)[https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L2195-L2197] to 5, so make sure your using the newest code and it should hopefully get fixed. Could you try it again and upload the serial log? If it still doesn't manage to finish the print, then I'll probably decrease the sending limit to 1 command at a time.

LogicalChaos, 2016-01-16:
Ah, OK.  I was just about to upload a failed log from a release code print.  Your description meshes nicely with what I see in the log.
```gcode
2016-01-14 09:44:03,959 - SERIAL - DEBUG - Recv: ok 22343
2016-01-14 09:44:03,969 - SERIAL - DEBUG - Send: N22344 G1 X15.933 Y77.845 E0.1317 F1620*50
2016-01-14 09:44:03,977 - SERIAL - DEBUG - Recv: wait
2016-01-14 09:44:03,980 - SERIAL - DEBUG - Send: N22345 G1 X15.933000 Y76.245000 E0.1317 F1620**18
2016-01-14 09:44:03,986 - SERIAL - DEBUG - Recv: T:234.0909 U:0.5000
2016-01-14 09:44:03,999 - SERIAL - DEBUG - Recv: ok 22344
```
I'll pull head and try again.  Thanks.

LogicalChaos, 2016-01-16:
Huh, short print just did it also, but a slightly different signature.  The M3D went into sending only waits during the print.  Canceled the print, still getting waits from the printer.

[wait.txt.zip](https://github.com/donovan6000/M3D-Fio/files/93051/wait.txt.zip)

LogicalChaos, 2016-01-18:
Really, really close.  It made through the print, but didn't process the last 4 commands, which makes sense with the logs.  Easy workaround (dup commands), but...  Last few commands.
```gcode
G1 X17.456 Y79.887 E2.7037
G92 E0
G1 E-2.5000 F1800
G91 ; relative positioning
G1 Z2 F90 ; move Z off print
M106 S0 ; turn fan off
M104 S0 ; turn extruder off
M18 ; turn motors off
```
Associated log.
```
2016-01-17 22:15:12,200 - SERIAL - DEBUG - Send: N305905 G1 X17.456000 Y79.887000 Z95.311635 E2.7037 F1728**122
2016-01-17 22:15:12,217 - SERIAL - DEBUG - Recv: ok 305901
2016-01-17 22:15:12,221 - SERIAL - DEBUG - Send: N305906 G92 E0*126
2016-01-17 22:15:12,234 - SERIAL - DEBUG - Recv: ok 305902
2016-01-17 22:15:12,238 - SERIAL - DEBUG - Send: N305907 G1 E-2.5000 F1800*46
2016-01-17 22:15:12,407 - SERIAL - DEBUG - Recv: ok 305903
2016-01-17 22:15:12,411 - SERIAL - DEBUG - Send: N305908 G91*38
2016-01-17 22:15:12,422 - SERIAL - DEBUG - Recv: ok 305904
2016-01-17 22:15:12,426 - SERIAL - DEBUG - Send: N305909 G1 Z2 F90*57
2016-01-17 22:15:12,441 - SERIAL - DEBUG - Recv: ok 305905
2016-01-17 22:15:12,444 - SERIAL - DEBUG - Send: N305910 M106 S0*89
2016-01-17 22:15:12,675 - SERIAL - DEBUG - Recv: T:229.7718 U:1.0000
2016-01-17 22:15:13,275 - SERIAL - DEBUG - Recv: ok 305906
2016-01-17 22:15:13,280 - SERIAL - DEBUG - Send: N305911 M104 S0*90
2016-01-17 22:15:13,291 - SERIAL - DEBUG - Recv: ok 305907
2016-01-17 22:15:13,295 - SERIAL - DEBUG - Send: N305912 M18*38
2016-01-17 22:15:13,307 - SERIAL - DEBUG - Recv: ok 305908
2016-01-17 22:15:14,015 - SERIAL - DEBUG - Changing monitoring state from 'Printing' to 'Operational'
```

donovan6000, 2016-01-18:
I've now limited the amount of commands that can be simultaneously sent to the printer to 1. So all the commands should get processed now.

LogicalChaos, 2016-01-19:
Will that have an impact on 'on the fly processing'?  Wasn't there a zits problem earlier?  Maybe unrelated though.

donovan6000, 2016-01-19:
It shouldn't cause any issues with on the fly pre-processing. It already has the new command ready for sending so there wont be much latency between when a confirmation is received and the next command is sent. 

The zits were most likely caused by commands being sent out of order and having too many G4 commands being sent. Both of those problems have been fixed.

LogicalChaos, 2016-01-20:
Works perfectly.  Thanks.

closed 2016-01-20 by LogicalChaos