donovan6000/M33-Fio #45: firmware update issue

georgestarcher, 2016-01-15:
When I connect a new m3d I get a request to update firmware to 2016010812. I then had to disconnect from USB to raspi/octoprint, power cycle printer before it would attempt to communicate again with printer. Then it would request to calibrate the bed. Calibration was slow then crashed completely. Moving the m3d back to mac laptop with m3d software it listed it as incompatible firmware and reverts it to 2015102303 before the printer works again. Octoprint is latest rpi image, then it updates itself and then I installed the m3d plugin. Pi is a B+. Maybe that is my issue and I should move it to a Pi2.

donovan6000, 2016-01-15:
Make sure you have the [Linux udev rule installed](https://github.com/donovan6000/M3D-Fio/blob/master/installers/Linux/90-m3d-local.rules) which ensures correct read/write access to the printer and prevents programs like ModemManager from messing up communicating with the printer.

Some people have had issues with running some of M3D Fio's longer operations, like updating firmware and calibrating the bed, which resulted from problems with their network's or proxy's configuration. I've never used OctoPrint on a network, so I don't have any experience with debugging this. I've also never ran OctoPrint on a Raspbery Pi, so I don't have any experience there either.

M3D's official software will always say the the firmware is incompatible if it's not the version that it supplies, so it makes sense that it reverted back to V2015102303. M3D Fio has stayed up to date with the changes to M3D's official software, so it currently requires firmware V2015122112 or later.

Can you upload OctoPrint's log so I can see more of what's happening? It's located in ~/.octoprint/logs/octoprint.log. What exactly happens after M3D Fio tries to update the firmware and when the bed calibration completely crashes?

georgestarcher, 2016-01-15:
I'll recreate it this weekend. I've moved over to an rpi2 and reimaged from the octopi distro. For now I can at least revert and work with the m3d software. I did happen to grab the code stream when I was trying to get it to give me the bed calibrate after firmware update. Not the log but I'll redo that this weekend.

[code.txt](https://github.com/donovan6000/M3D-Fio/files/91316/code.txt)

donovan6000, 2016-03-15:
I'm probably going to close this issue in the next couple of days since no ones responded to it in a few months. Let me know if you still have any interest in testing/solving it.

ThijsRoumen, 2016-03-16:
well I am not sure if it is related. I had a M3D which connected through M3DFio just fine. Now I have a new M3D (destroyed the previous one in some accident) which appears to have other firmware. It seems that the M3D software wants a more modern firmware on the M3D than the M3DFio software. When I connect to M3D I have to change the firmware and whenever I connect back to M3DFio this happens again. This prevents my server from auto-connecting.

donovan6000, 2016-03-16:
@ThijsRoumen The newest M3D Fio, V0.29, provides a modified version of M3D's newest official software, V2016012501, that increases the max temperature from 285 degrees to 315 degrees. I've given that modified firmware the version number V2016012503 so that M3D Fio will know that it has a newer firmware available which triggers it can ask the user if they want to update.

M3D's official software is only ever compatible with one firmware version at a time, and it'll require that you to upgrade/downgrade to the version that comes with their software. Since 2016012503 > 2016012501 M3D's official software will require that you downgrade in order to continue using their software.

M3D Fio is currently compatible with all firmwares >= 2015122112, so you can decline updating the firmware to V2016012503 and still be able to use it if your printer's current firmware falls into that range.

closed 2016-03-18 by donovan6000