donovan6000/M33-Fio #47: After print, printer try to go down scratching it

thibp, 2016-01-16:
Hello,
I noticed that with the latest version, for a tall print, that at the end it will try to go down.
Is this something know or to be fixed?
Thanks in advance

npag, 2016-01-22:
It doesn't even have to be a tall print. It happened to me before with a 3cm tall print.
After the printer printed all layers, the print head went up 1-2 cm and went right down again. The print got damaged because the hot nozzle melted a dent in the last position the last layer finished.

donovan6000, 2016-01-22:
This should be fixed in the [devel branch](https://github.com/donovan6000/M3D-Fio/tree/devel).

After a print is completed, M3D Fio sends several commands to the printer to do things like turn off the fan, turn off the heater, and move the extruder out of the way. For some reason a couple of these commands can be ignored by the printer, and this can cause the extruder to go down if the relative mode command, G91, is ignored. The devel branch doesn't rely on the relative command any more for moving the extruder after a print is done, so this issue is hopefully resolved.

npag, 2016-01-23:
I tried to install the dev version but I encountered a problem while uninstalling the previous version via the Plugin Manager. When I try to delete the plugin it fails due to insufficient permissions. So I did it manually via SSH. After I deleted the plugin successfully and restarted the server, the plugin was still in the list of installed Plugins. I tried to uninstall it a second time but it said that the plugin is not installed anymore, so I installed the dev version via SSH.
No clue if I am using the master or dev version right now but after printing, the print head rises a few cm and stays still. I can't issue commands to the printer until I disconnect and connect to it or powercycle it. There are lots of commands running in the Terminal while the print head is not doing anything. Heater and fan are both turned on in this state.

How can I determine which version I have installed right now? 
Octroprint drives me crazy sometimes... Shutdown commands are ignored very often, Plugin Manager sometimes finds no pip installation (or no permission error), server not reachable while doing complicated stuff.... It really annoys me.

donovan6000, 2016-01-26:
Just released M3D Fio V0.25 that includes all the changes in the devel branch, so just update the new newest release to see if this issue is fixed.

thibp, 2016-01-26:
Thanks. Will do as soon as I can.

npag, 2016-02-19:
This issue stopped for me after updating, all prints are undamaged by the print head. Seems to be fixed, thanks!

closed 2016-02-19 by donovan6000