donovan6000/M33-Fio #48: 1.3.0 can't print test border

ra100, 2016-01-16:
There are maybe more problems together. I ran calibrations and wanted to printe test border. I've got this from log:
```
2016-01-16 19:51:48,868 - octoprint.settings - WARNING - Could not convert '' to a valid integer when getting option []
2016-01-16 19:51:48,926 - octoprint - ERROR - Exception on /api/plugin/m3dfio [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev810+g3b35d92-py2.7.egg/octoprint/server/util/flask.py", line 709, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev810+g3b35d92-py2.7.egg/octoprint/server/api/__init__.py", line 97, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.24.3-py2.7.egg/octoprint_m3dfio/__init__.py", line 1349, in on_api_command
    self.sendCommands(self.getSaveCommands())
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.24.3-py2.7.egg/octoprint_m3dfio/__init__.py", line 4052, in getSaveCommands
    packed = struct.pack('f', softwareBackLeftOffset)
error: required argument is not a float
2016-01-16 19:51:48,951 - tornado.access - ERROR - 500 POST /api/plugin/m3dfio (192.168.150.11) 108.64ms
2016-01-16 19:51:59,051 - octoprint.plugin - ERROR - Error while calling plugin m3dfio
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev810+g3b35d92-py2.7.egg/octoprint/plugin/__init__.py", line 242, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.24.3-py2.7.egg/octoprint_m3dfio/__init__.py", line 2707, in on_event
    self.disableSleep()
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.24.3-py2.7.egg/octoprint_m3dfio/__init__.py", line 6113, in disableSleep
    bus = dbus.SessionBus()
  File "/usr/lib/python2.7/dist-packages/dbus/_dbus.py", line 211, in __new__
    mainloop=mainloop)
  File "/usr/lib/python2.7/dist-packages/dbus/_dbus.py", line 100, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
DBusException: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
2016-01-16 19:51:59,155 - tornado.access - WARNING - 404 GET /api/files/local/test_border (192.168.150.11) 67.60ms
2016-01-16 19:52:02,730 - tornado.access - WARNING - 404 GET /api/files/local/test_border (192.168.150.11) 56.82ms
2016-01-16 19:52:02,941 - tornado.access - WARNING - 404 GET /api/files/local/test_border (192.168.150.11) 57.81ms
```

I see one problem might be with test_border extension? `404 GET /api/files/local/test_border `.
And another that I don't start display manager on my RaPi, it's just in console.

ra100, 2016-01-17:
Dbus is fixed.
and `2016-01-16 19:52:02,941 - tornado.access - WARNING - 404 GET /api/files/local/test_border (192.168.150.11) 57.81ms` is just a warning, I thought that caused problem with popups, but it was something else. Both @donovan6000 already fixed. So closing this issue, and #49 should be closed too.

closed 2016-01-17 by ra100

donovan6000, 2016-01-17:
Yeah, I just added the exception handling that you did in the quick fix for the DBus issue. Also enforces data types more 'error: required argument is not a float' error.

Also just pushed some new code to fix the error 404 warning. In OctoPrint V1.3.0, the Gcode viewer cant analyze and display the print if the file being printed doesn't have a valid file extension and it'll generate the 404 warning that you get. I wasn't giving the test border and backlash cylinder file extensions because I wanted to make sure that they didn't overwrite any existing files that had the same name when I moved them to the upload folder and to prevent the user from printing an already processed calibration file if they aren't using on the fly pre-processing. But since that wont work anymore, I now have it so that they get .gcode extensions which prevents overwriting existing files since uploaded gcode files will always get a .gco extension. And they get deleted after the print fails, succeeds, or is cancelled to prevent the reprinting issue.

ra100, 2016-01-17:
I just tried it, start print and cancel, after canceling I get error (with on-thefly and without). But that is related to historyplugin. It probably wants to could filament usage, but file is already deleted. So that .gcode fix works fine, I get no tornado warnings or errors.
```
2016-01-17 02:34:57,039 - octoprint.events - ERROR - Got an exception while sending event MetadataStatisticsUpdated (Payload: {'path': u'test_border.gcode', 'storage': 'local'}) to <bound method Printer._on_event_MetadataStatisticsUpdated of <octoprint.printer.standard.Printer object at 0x739db150>>
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev810+g3b35d92-py2.7.egg/octoprint/events.py", line 131, in _work
    listener(event, payload)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev810+g3b35d92-py2.7.egg/octoprint/printer/standard.py", line 164, in _on_event_MetadataStatisticsUpdated
    self._selectedFile["sd"])
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev810+g3b35d92-py2.7.egg/octoprint/printer/standard.py", line 696, in _setJobData
    date = int(os.stat(path_on_disk).st_mtime)
OSError: [Errno 2] No such file or directory: '.octoprint/uploads/test_border.gcode'
```