donovan6000/M33-Fio #49: Latest commit will not print at all

LogicalChaos, 2016-01-16:
Commit: 083fde30a0b9c7df37ad60e5d0aa8e2b2839ad06

I can send manual commands, but when printing, it fails.  Octoprint log is showing an error calling the m3dfio plugin.

```
2016-01-16 13:06:56,621 - octoprint.plugin - ERROR - Error while calling plugin m3dfio
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.8-py2.7.egg/octoprint/plugin/__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint_M3DFio-0.24.3-py2.7.egg/octoprint_m3dfio/__init__.py", line 2680, in on_event
    if not self.disableSleep() :
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint_M3DFio-0.24.3-py2.7.egg/octoprint_m3dfio/__init__.py", line 6113, in disableSleep
    bus = dbus.SessionBus()
  File "/usr/lib/python2.7/dist-packages/dbus/_dbus.py", line 211, in __new__
    mainloop=mainloop)
  File "/usr/lib/python2.7/dist-packages/dbus/_dbus.py", line 100, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
```
[serial_fail.log 2.zip](https://github.com/donovan6000/M3D-Fio/files/93072/serial_fail.log.2.zip)
[octoprint.log.zip](https://github.com/donovan6000/M3D-Fio/files/93073/octoprint.log.zip)

LogicalChaos, 2016-01-16:
I fell back to commit 0c65186ef899d22eacbc1739aa7e52f573751b28, left the firmware 2016010812 and it's working so far.  I didn't try any other commits.

ra100, 2016-01-16:
Seem the same as https://github.com/donovan6000/M3D-Fio/issues/48. I think the problem is in sleep preventing feature which can probably work only when you have display manager running.

LogicalChaos, 2016-01-16:
I look at that #48, but obviously not close enough.

ra100, 2016-01-16:
So here is quick workaround https://github.com/ra100/M3D-Fio/commit/e99ca6a4aaa9b7d9a0dd22b5e41d447e617159e3

donovan6000, 2016-01-17:
Should be fixed now. Have the DBus session initialization fail would propagate and cause a lot of things to fail.

LogicalChaos, 2016-01-17:
Don't receive those errors in octoprint.log anymore, but it's still not printing (9d2aa7b4d3cc8c82e621a9e9d0e5c08fcc7469f8).  Here is printing the border.
```gcode
2016-01-16 19:15:20,999 - SERIAL - DEBUG - Changing monitoring state from 'Connecting' to 'Operational'
2016-01-16 19:15:21,037 - SERIAL - DEBUG - Send: M110
2016-01-16 19:15:22,013 - SERIAL - DEBUG - Send: M110
2016-01-16 19:15:22,015 - SERIAL - DEBUG - Send: M105
2016-01-16 19:15:22,019 - SERIAL - DEBUG - Recv: ok
2016-01-16 19:15:22,021 - SERIAL - DEBUG - Recv: T:0
2016-01-16 19:15:22,143 - SERIAL - DEBUG - Recv: wait
2016-01-16 19:15:42,029 - SERIAL - DEBUG - Changing monitoring state from 'Operational' to 'Printing'
2016-01-16 19:15:42,544 - SERIAL - DEBUG - Send: N0 M110 N0*125
2016-01-16 19:15:43,064 - SERIAL - DEBUG - Recv: wait
2016-01-16 19:16:16,928 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:17,925 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:18,919 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:19,917 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:20,912 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:21,909 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:22,906 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:23,901 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:24,896 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:28,879 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:30,871 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:31,865 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:32,864 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:33,858 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:34,853 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:35,848 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:37,843 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:38,838 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:39,833 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-01-16 19:16:40,829 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
```

LogicalChaos, 2016-01-17:
Commit 57c3e2cca558065710726b4289a3614c04c948da is what breaks the build for me.

donovan6000, 2016-01-17:
Do you ever see this dialog box? I added the ability to set the filament temperature and type right before printing when using on the fly pre-processing. It'll stall the print until that dialog is responded to.
[image]

LogicalChaos, 2016-01-17:
I was just going to comment on that.  I don't see that dialog with 57c3e2cca558065710726b4289a3614c04c948da, but I do with db1538cae29b95a2b54a6251d60d1eb5188642e2.

LogicalChaos, 2016-01-17:
And as a request, can that dialog be disabled?  I have the temperature in the gcode files and don't need to be asked again :-)  Maybe configurable?

donovan6000, 2016-01-17:
I did some big changes last night to the way that M3D Fio displays dialogs, so it's still kinda buggy.

That dialogs optional now. There's a setting underneath the pre-processor on the fly called 'Change settings before print'. Also, I think you already know this, but make sure you have thermal bonding disable if you want to use the temperature commands that in the gcode file. Otherwise those commands get filtered out.

LogicalChaos, 2016-01-17:
Thanks, it's working now without the dialog.  And yeah, you have good tool-tips for the thermal bonding, so it was rather obvious I didn't want that enabled :+1:

closed 2016-01-17 by LogicalChaos