donovan6000/M33-Fio #50: install.sh script installs newest (unstable) version

drschlaumeier, 2016-01-18:
wget https://github.com/donovan6000/M3D-Fio/archive/master.zip

downloads the newest version but not the latest stable version...
why not using

wget https://github.com/donovan6000/M3D-Fio/archive/0.24.3.zip

ra100, 2016-01-18:
Or maybe create devel branch? So in master is always latest stable version.

drschlaumeier, 2016-01-18:
> So in master is always latest stable version.
Are you sure? When I do "pip install https://github.com/donovan6000/M3D-Fio/archive/master.zip" I get all the latest commits installed ... probably untested?
I mean when I do the intall today I get something different when I do the intall tomorrow...

donovan6000, 2016-01-18:
Ok, devel branch has been created.

Some new code was just pushed to the master branch that could make it more stable. I've never done any M3D Fio testing on a network or on a Raspberry Pi, and I feel like a majority of issues are related to how those things are configured.

LogicalChaos, 2016-01-19:
FYI:  I'm only using a Raspberry Pi, OctoPi based.  Maybe that was apparent from other issues :-)

closed 2016-01-26 by donovan6000