donovan6000/M33-Fio #51: Installing M3D-Fio from repository -> Plugin homepage is set to exploitkings

JollyOrc, 2016-01-19:
[snippet.txt](https://github.com/donovan6000/M3D-Fio/files/96350/snippet.txt)
See attached snippet.txt - that is the html code displayed for M3D-Fio in the plugin list after having installed the plugin from the OctoPrint Plugin repository.

JollyOrc, 2016-01-19:
(Mind: All other plugins point to the relevant github pages. And the URL doesn't exactly inspire confidence.. This is maybe even legitimate, but it certainly doesn't look that way, and you can't find anything on the plugin on that page)

donovan6000, 2016-01-19:
Exploit Kings is one of my other projects. Like it says when you visit it, "This site is still being created. Please ignore it for now." M3D Fio is copyrighted under Exploit Kings and my [email listed in the code](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L6) uses the Exploit Kings domain name, so there's nothing to worry about.

I see your point though. A plugin's homepage should be somewhere that actually has information about the plugin, so I've changed its homepage to the github page in the [devel](https://github.com/donovan6000/M3D-Fio/tree/devel) branch.

closed 2016-01-26 by donovan6000

JollyOrc, 2016-01-27:
great, thanks for that!