donovan6000/M33-Fio #52: Need to set temperature 50° higher than usual

ra100, 2016-01-21:
I've got this issue a long time, maybe more than month. I need to set PLA temperature to 260°C to make it print, so pla it doesn't get stuck in printhead. Before I've been able to print from 190°C.
I already contacted m3d support and they responded that it's OK, to set the temperature to value that works for me. But I'm not sure what caused this change. Could this be extrude current issue? I've always had 500mA setting and it didn't change, but it keeps me thinking if I shouldn't try 660mA, but I don't want to damage heater. I also wanted to know from m3d which extruder I have, but there is no way to know, as they said.

donovan6000, 2016-01-21:
I was having a similar problem lately where it seems like barley any to no filament was being extruded. I usually don't have filament loaded when I'm developing for M3D Fio cause I don't want to waste a lot of filament, so I thought that doing so many dry prints might have damaged the nozzle somehow. I ended up [removing the nozzle](https://printm3d.com/solutions/article.php?id=22) and blasting everything with compressed air in case any debris was hindering the filaments path. It's working better now, so whatever I did fixed it. So it might be worth doing the same to your printer.

I wish M3D was more open about their heating system. All I know is that if the beginning of your printer's serial number starts with [one of these values](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L3154), then it's using a 500mA heater, however I don't know if not being in that list means that it's using a 660mA heater. The official software has never changes that list since June, so I think all newer printer are using a 660mA heater. I also think using the incorrect extruder current settings would damage something, so I wouldn't recommend that. I did look at the heating calibration settings in your EEPROM a while ago, and they were very similar to mine. I also have a 500mA heater, so I'm feeling like your problem isn't a current issue or a heater calibration issue. Do your prints look gloopy or over extruded at all like it would if the temperature was really 260°C?

I saw [this video](https://www.youtube.com/watch?v=f0VpAZhdNmg) a while back about placing aluminum foil between the heater and nozzle to increase the heat flow. Maybe your heater moved slightly out of place and it's not contacting the nozzel as much now.

ra100, 2016-01-21:
Thanks. I'll try to experiment a bit, I've already changed nozzle for spare I ordered with printer. And results were same as before. I'll try to clean them properly and I'll see what it does. Now I'm printing at 240°C. At 260, print was too much melted.

LogicalChaos, 2016-01-22:
I print PLA at 230 (Hatchbox and MakerGeeks).  You should try one of the [temp calibration towers](http://www.thingiverse.com/thing:915435).  I've included the one I customized, it goes from 245 to 195 in 5 degrees increments.  MakerGeeks all come in good (print layers don't flake apart) at 210+,  a maroon one (forgot the make) at 220+, Hatchbox 200+ for black and 210+ for blue.  Highly recommend printing them.

[tour_de_chauffe_v2_20151025-12423-gj7053-0.gcode.zip](https://github.com/donovan6000/M3D-Fio/files/100287/tour_de_chauffe_v2_20151025-12423-gj7053-0.gcode.zip)

ra100, 2016-01-22:
For now I cleaned nozzle with compressed air and tried to add aluminium foil under heater. Seem like it works a bit better, but 210°C seems lowes I can go to see filament extrude in "normal" speed. I was just trying to set temperature to 240, 230, 220, 210, 200 and then extrude 10mm of filament and watched how long extruded filament was. Under 205 it just curls around nozzle tip.
Next I'll try calibration towers.

ra100, 2016-01-24:
Temp calibration tower didn't work well.
As I'm looking at serial numbers, mine starts with BK15042901100 (and I got my printer in May), but that number is not in list. So I'm not sure what to take from that.

donovan6000, 2016-01-28:
I just ordered an AVR Dragon, so I'll be able to see how their heating system works in a couple of days by doing some in-circuit debugging. Do you have a way of measuring the actual temperature of the heater to see if the calibration is incorrect?

The printer has two heater settings in it's EEPROM, heater temperature measurement B and heater resistance M. I mentioned before that our values are similar, and assuming their stored in IEEE single precision floating point representation in little endian, they are:

Heater temperature measurement B
Mine: -409.0596618652344
Yours: -394.3919677734375

Heater resistance M
Mine: 246.4896240234375 (My heater's resistance at room temperature is about 2.5‎Ω, but I doubt that that's related to this value)
Yours: 276.1018981933594

Any idea about what they mean?

ra100, 2016-01-28:
I'm affraid I have nothing to measure heater temperature with.
And those values, maybe temperature in Fahrenheit? 409°F is  209°C, 394 -> 201, but I don't know it makes any sense.
And I'm not sure what else I could do.

LogicalChaos, 2016-01-29:
> On Jan 24, 2016, at 7:58 AM, Rastislav Švarba <notifications@github.com> wrote:
> 
> Temp calibration tower didn't work well.

Huh.  What happened?  Picture?

LogicalChaos, 2016-01-29:
> On Jan 28, 2016, at 3:52 AM, Rastislav Švarba <notifications@github.com> wrote:
> 
> I'm affraid I have nothing to measure heater temperature with.

Has anyone tried an IR thermometer to see if it reads anything accurately?  I may have to take mine apart a bit...

ra100, 2016-01-29:
> Huh.  What happened?  Picture?
I'm sorry, I don't have a picture, but it ended up that way it printed brim nad when it started to print tower it not at all sticked to the printbed. Maybe wrong offsets calibrations.

ra100, 2016-01-31:
And this is how my test borders look like on different temperatures after full bed calibration.
[image]
[image]

ra100, 2016-02-29:
Ok, this seems like HW problem, will close this. Thanks wveryone for your help.

closed 2016-02-29 by ra100

donovan6000, 2016-04-21:
Finally got around to looking into how the printer's heating works. The equation they're using to convert between the ADC readings and actual temperatures is `Temperature = ADC Value * Heater resistance M + Temperature measurement B`

Temperature measurement B is applied as an offset, so you can adjust that value to compensate for the additional 50 degrees. Changing yours from -394.3919677734375 to -444.3919677734375 will change all the readings to be 50 degrees less.

So all you have to do is change the Temperature measurement B value in the EEPROM from 2C 32 C5 C3 to 2C 32 DE C3.
[image]

TonyLHansen, 2016-07-08:
Do they alternate between putting current through the heater coils and measuring the resistive value? If so, what is the duty rate of that cycle?

donovan6000, 2016-07-08:
They measure the resistance while they heat it at the same time. Measuring the resistance while it's not being heated only tells us if the heater is connected or not, which they only do when the printer it first powered on to determine if the heater is broken or not.

The actual equation that they use to calculate the temperature is this. 
`temperature =  (-245 / -450 * 4) * ((pin A3 voltage - pin A4 voltage) / 1.1 * 2048 * / 2) / (pin A5 voltage / 1.1 * 4096) * eepromHeaterResistanceM + eepromHeaterTemperatureMeasurementB`

The microcontroller has an internal bandgap voltage of 1.1V that can be used as a reference for the ADC controller. The signed 12-bit `pinA3 - pinA4` and unsigned 12-bit `pinA5` are with reference to that bandgap, which is whey there's the `1.1`, `2048`, and `4096` comes from.

The `/ 2` comes from them using a 1/2x gain on the ADC reading.

The compiler they're using evaluates all the constants at compile time, so the `-245 / -450 * 4` is just how I think the compiled 2.177777778 value was calculated. 245 being the ideal eepromHeaterResistanceM and -450 being the ideal eepromHeaterResistanceM. I have no proof of this though, and I have no idea what those values mean anyway.

TonyLHansen, 2016-07-08:
Thanks for the info. Umm, Can you correct the "* / 2"? Is it a multiply or divide?

donovan6000, 2016-07-08:
Oops, it should be / 2

nujugi, 2017-05-09:
I n00d to this.
But I have have my M3D @ 280c to get half disent print.
How do I get to edit the EEPROM?

donovan6000, 2017-05-10:
@nujugi The formula for the temperature is `Temperature = ADC Value * Heater resistance M + Temperature measurement B`, so you can change the `Temperature measurement B` setting to apply a fixed offset to the temperature reading.

In the EEPROM section of OctoPrint's Control tab edit the value starting at address 0x2BA to be the desired offset lower than it currently is. The values is a single precision floating point number stored in little endian. So if yours is `A3 87 CC C3` then you can use [this tool ](https://gregstoll.dyndns.org/~gregstoll/floattohex/) to convert that value in to big endian notation, `C3 CC 87 A3`, to a floating point number. For this example, the `A3 87 CC C3` value is -409.06. If you want your temperature readings to be 50 degrees lower than that then you would want that value to be -459.06, which is `C3 E5 87 AE` in big endian. So the value you want in EEPROM is `AE 87 E5 C3`.

nujugi, 2017-05-10:
Thanks for fast reply.
But i don't get a number like -409 out of the tool.
`1E 74 CE C3` Float value: 1.296e-20?

nujugi, 2017-05-10:
Aha.. I see read the hex from the right :P