donovan6000/M33-Fio #53: Double movement when manually calibrating

ra100, 2016-01-24:
While manual calibration or during complete calibration in manual phase the head moves sometimes twice to the center.
Here is head movement: position: Center, Back, Front, Left, Right
`B->R->C->FL->B->R->C->B->R->C->FR->B->R->C->BR->B->R->C->B->R->C->BL`
See, there is double sequence B->R->C between FL and FR and BR and BL positions.
When using on-the-fly settings is a bit different:
`B->R->C->FL->FR->B->R->C->BR->BL`
(`B->R->C` sequence is obviously movement to "reset" positions and move to center)

donovan6000, 2016-01-26:
Should be fixed now.

I noticed that some commands would occasionally get ignored when several commands were sent as once to the printer by any of the buttons, so I added G4 commands and delays in M3D Fio in between sends which helped. However this also caused the double homing that you see since the newer printer firmwares seem to do that if they get sent G28 G4 fast enough :/ So it now doesn't send a G4 command before a G28 command which solved this problem and still prevents commands from being ignored.

ra100, 2016-01-26:
Now it really works nicely, without useless homing.

closed 2016-01-26 by ra100