donovan6000/M33-Fio #54: M3D stops printing after ~6h

Finkes, 2016-01-25:
I have beendete printing a relative big model for over 6h but then the printer just stops moving. It is steal heating and octoprint seams to send the commands to the printer but for some reason the printer ignores those commands. I tried it another and again after more then 6h the same issue happened. However this time the printer printer the model a bit longer then the first time. Does someone know what could be going in here? I know that this might be an issue with the printer itself or with octoprint but since octoprint is not supported by the official m3d team, this project is my only hope for a solution.

donovan6000, 2016-01-26:
Try out the newest release, V0.25. This issue might be related to #43 which got fixed.

Finkes, 2016-01-27:
It's working now with V0.25. Super awesome :-) thanks! Keep it going!

closed 2016-01-27 by Finkes