donovan6000/M33-Fio #57: Cannot install v0.25 on frest octoprint install

tenitram, 2016-01-27:
Hi. I cannot install v0.25 neither from command line nor from web interface.

Please find attached steps:
```
root@odroid:~/OctoPrint# ./venv/bin/octoprint --version
OctoPrint version 1.2.8.post0.dev0+gdedadbc
root@odroid:~/OctoPrint# /root/OctoPrint/venv/bin/pip install https://github.com/donovan6000/M3D-Fio/archive/master.zip
Downloading/unpacking https://github.com/donovan6000/M3D-Fio/archive/master.zip
  Downloading master.zip (4.8Mb): 4.8Mb downloaded
  Running setup.py egg_info for package from https://github.com/donovan6000/M3D-Fio/archive/master.zip
    
Cleaning up...
root@odroid:~/OctoPrint# service octoprint restart
Restarting Octoprint Daemon: Octoprint.
```
After restart there is no M3D Fio menu under settings. But when I install V0.24.3 then plugin installs correctly:
```
root@odroid:~/OctoPrint# /root/OctoPrint/venv/bin/pip install https://github.com/donovan6000/M3D-Fio/archive/V0.24.3.zip
Downloading/unpacking https://github.com/donovan6000/M3D-Fio/archive/V0.24.3.zip
  Downloading V0.24.3.zip (unknown size): 4.6Mb downloaded
  Running setup.py egg_info for package from https://github.com/donovan6000/M3D-Fio/archive/V0.24.3.zip
    
Installing collected packages: OctoPrint-M3DFio
  Found existing installation: OctoPrint-M3DFio 0.25
    Not uninstalling OctoPrint-M3DFio at /usr/local/lib/python2.7/dist-packages, outside environment /root/OctoPrint/venv
  Running setup.py install for OctoPrint-M3DFio
    
Successfully installed OctoPrint-M3DFio
Cleaning up...
root@odroid:~/OctoPrint# service octoprint restart
Restarting Octoprint Daemon: Octoprint.
```

donovan6000, 2016-01-28:
I've always encountered weird errors when installing any OctoPrint's plugin on Linux, and trying to install them several times will usually succeed. 
```
donovan6000@dv6-6b22he:~/Programming/Computers/M3D Fio$ sudo pip install https://github.com/donovan6000/M3D-Fio/archive/master.zip
[sudo] password for donovan6000: 
Downloading/unpacking https://github.com/donovan6000/M3D-Fio/archive/master.zip
  Downloading master.zip (unknown size): 4.8MB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 304, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1198, in prepare_files
    do_download,
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1376, in unpack_url
    self.session,
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 572, in unpack_http_url
    download_hash = _download_url(resp, link, temp_location)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 433, in _download_url
    for chunk in resp_read(4096):
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 421, in resp_read
    chunk_size, decode_content=False):
  File "/usr/share/python-wheels/urllib3-1.9.1-py2.py3-none-any.whl/urllib3/response.py", line 256, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/share/python-wheels/urllib3-1.9.1-py2.py3-none-any.whl/urllib3/response.py", line 186, in read
    data = self._fp.read(amt)
  File "/usr/lib/python2.7/httplib.py", line 611, in read
    s = self.fp.read(amt)
  File "/usr/lib/python2.7/socket.py", line 384, in read
    data = self._sock.recv(left)
  File "/usr/share/python-wheels/urllib3-1.9.1-py2.py3-none-any.whl/urllib3/contrib/pyopenssl.py", line 188, in recv
    data = self.connection.recv(*args, **kwargs)
ZeroReturnError

Storing debug log for failure in /home/donovan6000/.pip/pip.log
ddonovan6000@dv6-6b22he:~/Programming/Computers/M3D Fio$ sudo pip install https://github.com/donovan6000/M3D-Fio/archive/master.zip
Downloading/unpacking https://github.com/donovan6000/M3D-Fio/archive/master.zip
  Downloading master.zip (4.8MB): 4.8MB downloaded
  Running setup.py (path:/tmp/pip-kVIS0m-build/setup.py) egg_info for package from https://github.com/donovan6000/M3D-Fio/archive/master.zip
    
Installing collected packages: OctoPrint-M3DFio
  Running setup.py install for OctoPrint-M3DFio
    
Successfully installed OctoPrint-M3DFio
Cleaning up...
donovan6000@dv6-6b22he:~/Programming/Computers/M3D Fio$ 
```

You can also try uninstalling any previously installed version before hand just to make sure that it's not interfering with installer a newer version in any way.
`pip uninstall OctoPrint-M3DFio`

tenitram, 2016-01-28:
After deeper look I managed to install from master. The problem was that installing plugin was failing silently, even with verbose parameter. But in octoprints log file there was an info that dependency  was missing: dbus. After installing python-dbus I was able to properly install plugin. (previous version was working ok on the same environment) 

Thx for your hard work with the plugin :)

closed 2016-01-28 by tenitram