donovan6000/M33-Fio #58: During complete calibration head didn't move to FR and BL corner

ra100, 2016-01-28:
In manual calibration phase on complete bed calibration, head didn't move to Front Right and Back Left corner.

donovan6000, 2016-01-29:
This is a result of the printer ignoring some commands in a non repeatable way, and it's related to the double homing issue that we just fixed. I've tried a lot in the past to overcome this issue. So now it'll send the [absolute command G90 twice, which will actually get sent 4 times because M3D Fio already sends it twice, and the movement command G0 twice](https://github.com/donovan6000/M3D-Fio/blob/devel/octoprint_m3dfio/static/js/m3dfio.js#L5784-L5789). There's already G4 commands and delays that get sent in between sending commands, so hopefully all these things combined will fix this problem.

ra100, 2016-01-29:
Just tried it, works fine.

ra100, 2016-01-29:
Well almost, I just looked at settings, and bed offsets are set to 0.
serial.log https://gist.github.com/ra100/7a222cdf200ee1758d93

ra100, 2016-02-07:
bed offset values are being saved saved

closed 2016-02-07 by ra100