donovan6000/M33-Fio #59: Test border - No fan speed has been detected

ra100, 2016-01-28:
When printing test border I get `No fan speed has been detected in this file which could cause the print to fail.` but it's probably just my problem with HW. I couldn't get the fan to spin for some time now, already tried to change fan. And now, I've got external power supply for fan.

donovan6000, 2016-01-29:
It's fixed now. It wasn't taking into account if the preparation pre-processor was being used. It'll display that message whenever trying to print and no fan G-code commands are found in the file and none are added to it.

And M3D Fio now includes an option to set one of the Raspberry Pi's GPIO pins high at the start of the a print's fourth layer and low when a print is done or canceled. I included it so people could use external fans with a setup similar to [this](http://www.allaboutcircuits.com/projects/raspberry-pi-project-control-a-dc-fan/).

M3D's fan are really cheap though. Mine started to sound like a jet engine after a few weeks. I can't sleep in the same room as my printer, so I really doubt that a cat could. Also this image isn't even a fully assembled printer XD
[image]

ra100, 2016-02-06:
Thanks for explaining, so this really is not an issue :)

closed 2016-02-06 by ra100