donovan6000/M33-Fio #61: M3D will stop on Raspberry Pi

mdikov, 2016-01-31:
I tried several models that print without a problem on M3D default software. When I print via Octoprint on Raspberry Pi , the raft or the brim print fine than the printer will start the first layer and stop before it's done. The GCode viewer and terminal continue to send commands.

I also tried gcode generated on another Cura and the same happens.

tobiasbischoff, 2016-01-31:
Yes i have that too. But it seems model independent. I can try to send the exact same gcode to the printer - first time the m3d stopps at 60%, second time it justs prints thru fine.

mdikov, 2016-01-31:
Is it possible that the  serial connection drops? The interesting part is that the printer seems to be responding to the commands in terminal. Maybe a firmware or gcode incompatibility?

Sent from a mobile phone

totterfree, 2016-02-08:
I can confirm this same issue. I'm using the raspberry pi zero. Both the Pi and the M3D are connected to a reliable power source. The M3D is connected to the Pi via a usb hub which also has a wifi adapter connected. I have had the printer freeze on all of my prints so far. I will check for any terminal output on my next attempt

donovan6000, 2016-02-08:
I don't own or have access to a Raspberry Pi, and it's difficult for me to fix issues that I can't recreate. Can any of you provide an OctoPrint log of when this happens? It's located at ~/.octoprint/logs/octoprint.log.

tobiasbischoff, 2016-02-08:
Actually it has been quite some time that this happend to me. Since i upgraded to 25 it happened once, then never again. Although i have tons of other problems with the m3d now, like skipping, not sticking to bed etc, overall bad print quality. Its that much that i'm pretty much fed up with it.

mdikov, 2016-02-08:
I can provide my latest logs later in the week. 

Sent from a mobile phone

donovan6000, 2016-02-09:
Make sure your using M3D Fio's latest [devel build](https://github.com/donovan6000/M3D-Fio/tree/devel) when testing this issue so I can see if it hasn't been fixed yet. You can install using OctoPrint's plugin manager by installing a plugin from the following URL:
`https://github.com/donovan6000/M3D-Fio/archive/devel.zip`

mdikov, 2016-02-14:
Here you go. I started the model and got one or two layers then just
stopped.

mdikov, 2016-02-14:
Also to confirm I upgraded to 0.26 from the develop branch.

Here is what comes out of the terminal:

Recv: ok 1274
Send: N1275 G1 X60.799934 Y55.368109 Z0.366256 E60.247521**20
Recv: ok 1275
Send: N1276 G1 X61.582902 Y57.208478 Z0.372373 E60.276624**26
Recv: ok 1276
Send: N1277 G1 X62.431028 Y59.202000 Z0.378975 E60.30815**37
Recv: ok 1277
Send: N1278 G1 X62.365 Y58.195 E60.31230*111
Recv: ok 1278
Send: N1279 G1 X62.711028 Y58.195000 Z0.375589 E60.31230 F780**113
Recv: ok 1279
Send: N1280 G1 X62.365 Y43.770 E60.52221*106
Recv: ok 1280
Send: N1281 G1 X62.711028 Y54.195000 Z0.344275 E60.370507**16
Recv: ok 1281
Send: N1282 G1 X62.711028 Y52.195000 Z0.328618 E60.399611**19
Recv: ok 1282
Send: N1283 G1 X62.711028 Y50.195000 Z0.312962 E60.428715**19
Recv: ok 1283
Send: N1284 G1 X62.711028 Y48.195000 Z0.297305 E60.457818**16
Recv: ok 1284
Send: N1285 G1 X62.711028 Y46.195000 Z0.281648 E60.486922**16
Recv: ok 1285
Send: N1286 G1 X62.711028 Y43.770000 Z0.262664 E60.52221**45
Recv: T:208.9900
Recv: ok 1286
Send: N1287 G1 X67.465 Y43.770 E60.59642*101
Recv: ok 1287
Send: N1288 G1 X67.811028 Y43.770000 Z0.208668 E60.59642**35
Recv: ok 1288
Send: N1289 G0 F3000 X66.528 Y44.592*126
Recv: ok 1289
Send: N1290 G0 X66.528000 Y45.542000 Z0.225023 F3000**48
Recv: ok 1290
Send: N1291 G1 F780 X63.187 Y47.934 E60.66518*60
Recv: ok 1291
Send: N1292 G1 X63.187000 Y48.884000 Z0.286559 E60.66518 F780**113
Recv: ok 1292
Send: N1293 G0 F3000 X63.187 Y48.429*124
Recv: ok 1293
Send: N1294 G1 F780 X66.641 Y44.975 E60.73626*61
Recv: T:215.0885
Recv: ok 1294
Send: N1295 G1 X64.947242 Y47.014786 Z0.264390 E60.694283 F780**71
Recv: ok 1295
Send: N1296 G1 X66.987028 Y44.975000 Z0.226825 E60.73626 F780**119
Recv: ok 1296
Send: N1297 G0 F3000 X66.641 Y45.470*113
Recv: ok 1297
Send: N1298 G0 X66.987028 Y46.420000 Z0.230700 F3000**63
Recv: ok 1298
Send: N1299 G1 F780 X63.187 Y48.924 E60.80734*62
Recv: ok 1299
Send: N1300 G1 X65.226786 Y47.834214 Z0.256744 E60.765363 F780**66
Recv: T:208.9900
Recv: ok 1300
Send: N1301 G1 X63.187000 Y49.874000 Z0.294309 E60.80734 F780**112
Recv: ok 1301
Send: N1302 G0 F3000 X63.187 Y49.419*119
Recv: ok 1302
Send: N1303 M105*22
Recv: ok 1303
Send: N1304 G1 F780 X66.641 Y45.965 E60.87842*50
Recv: ok 1304
Send: N1305 G1 X64.947242 Y48.004786 Z0.272140 E60.836443 F780**69
Recv: ok 1305
[... 118 lines cut ...]
Recv: ok 1362
Send: N1363 G0 X66.987028 Y53.350000 Z0.284951 F3000**59
Recv: ok 1363
Send: N1364 G1 F780 X63.187 Y55.854 E61.80244*52
Recv: T:215.0885
Recv: ok 1364
Send: N1365 G1 X65.226786 Y54.764214 Z0.310995 E61.760463 F780**75
Recv: ok 1365
Send: N1366 G1 X63.187000 Y56.804000 Z0.348559 E61.80244 F780**120
Recv: ok 1366
Send: N1367 M117 [          ]   1%%*32
Recv: e1
Recv: T:208.9900
Recv: wait

=============this is where it stops ============
Send: N1368 G0 F3000 X63.187 Y56.349*119
Recv: T:208.4817
Recv: wait
Recv: T:215.8991
Recv: wait
Recv: T:215.8991
Recv: wait
Recv: T:215.8991
Recv: wait

donovan6000, 2016-02-14:
@mdikov The problem is that the M117 command is used to display stuff on an LCD screen, and M3D's firmware doesn't recognize that command so it returns an error, e1. M3D Fio waits for an 'ok' response before sending the next command, so it's never able to send the next command which causes the print to stop.

The latest change to the devel branch [adds M117 to list of commands that M3D Fio's validation pre-processor removes](https://github.com/donovan6000/M3D-Fio/commit/cec93d7addac7856f2243d0c686d4b2968300458), so it should be able to print now.

mdikov, 2016-02-14:
Great. I'll check it out. 

Just curious why would cura add this command? Perhaps there's a way turn this off in the cura profile. 

Sent from a mobile phone

donovan6000, 2016-02-14:
@mdikov The M117 command is probably in the Cura profile's start or end G-code, which is why it gets added in.

mdikov, 2016-02-16:
Thank you @donovan6000! Works great now.

closed 2016-02-16 by mdikov