donovan6000/M33-Fio #62: "You've been disconnected from the server." in v0.25 on RPi1 Model B

rondhi, 2016-01-31:
I'm using a Raspberry Pi Model B and an OctoPi image.
Upon installing M3D-Fio v0.25, I'm unable to log into OctoPrint at all because I get a message saying "You've been disconnected from the server. The browser will now refresh". Pressing OK refreshes the page, and this happens in an endless loop.

This issue seems to be introduced in v0.25, as v0.24.3 does not have this issue.

Let me know what I can do if you need more information.

LogicalChaos, 2016-01-31:
> On Jan 31, 2016, at 9:46 AM, rondhi <notifications@github.com> wrote:
> 
> I'm using a Raspberry Pi Model B and an OctoPi image
> Upon installing M3D-Fio v025, I'm unable to log into OctoPrint at all because I get a message saying "You've been disconnected from the server The browser will now refresh" Pressing OK refreshes the page, and this happens in an endless loop
> 
> This issue seems to be introduced in v025, as v0243 does not have this issue
> 
I'm running the same setup without any issues.  Have you tried closing the browser tab?  I hate to say it, but rebooted the Raspberry?  What browser/host?

rondhi, 2016-01-31:
Alright I seem to have fixed the issue. But I'm not sure what solved it. I reinstalled v0.25, but I also overclocked my RPi1 to the Turbo setting in raspi-config. Don't worry, I have a heatsink installed on the RPi.

closed 2016-01-31 by rondhi