donovan6000/M33-Fio #63: Can't install on OctoPi nightly 0.13.0

guysoft, 2016-01-31:
Hey,
I tried installing M3D Fio on the most recent Nightly today 2016-01-31_2015-11-21-octopi-jessie-lite-0.13.0.zip . Although the python package seems to install fine, I don't see any settings for the m3d, also the plugin is not listed in the plugin manager. I can search and install it again.

```
pi@octopi:~ $ /home/pi/oprint/bin/pip install https://github.com/donovan6000/M3D-Fio/archive/master.zip
Collecting https://github.com/donovan6000/M3D-Fio/archive/master.zip
  Downloading https://github.com/donovan6000/M3D-Fio/archive/master.zip (4.8MB)
    100% |████████████████████████████████| 4.8MB 11kB/s 
  Requirement already satisfied (use --upgrade to upgrade): OctoPrint-M3DFio==0.25 from https://github.com/donovan6000/M3D-Fio/archive/master.zip in ./oprint/lib/python2.7/site-packages
```

Am I missing something?

guysoft, 2016-01-31:
Hey, tried installing and installing, and saw I get this in the GUI: 
[image]

tenitram, 2016-01-31:
Had the same issue - in my case it was python-dbus package missing - check octoprint log file

donovan6000, 2016-02-07:
Can you install the devel branch? https://github.com/donovan6000/M3D-Fio/archive/devel.zip

guysoft, 2016-02-07:
ok @tenitram was right, I had to install python dbus on the virtualenv, which can't be done now by simple apt-get.
So what needs to be done is:

```
sudo apt-get build-dep python-dbus
/home/pi/oprint/bin/pip install -U git+https://github.com/posborne/dbus-python.git
```

donovan6000, 2016-02-07:
I just tested out installing and running the devel branch without having Python DBus installed and it worked without any issues, so the next release will be able to run without needing Python DBus installed.

closed 2016-02-07 by donovan6000