donovan6000/M33-Fio #64: Can't connect to printer using the Devel branch after commit ce847be29802dfa8f39d45498b7fea6811aa36ff , "Trying to improve connecting with heatbed"

DGdev91, 2016-02-05:
I can't connect  anymore to the printer using the devel branch after commit ce847be29802dfa8f39d45498b7fea6811aa36ff , "Trying to improve connecting with heatbed".

First i tought there was something wrong related to PySerial 3, since i'm currently using it on ArchLinux ARM (And there was, i'm releasing a fix) but the problem is still there.

Maybe the program is detecting an heatbead even if haven't got one?

donovan6000, 2016-02-07:
Do you happen to have any USB devices connected with a 0x1A86 VID and 0x7523 PID?

Try removing everything [inside](https://github.com/donovan6000/M3D-Fio/blob/devel/octoprint_m3dfio/__init__.py#L588-L686) the 'MonitorHeatbed' function and try connecting to see if it's really the heatbed that's causing the issue.

DGdev91, 2016-02-08:
my lsusb:
lsusb
Bus 001 Device 009: ID 18a5:0408 Verbatim, Ltd 
Bus 001 Device 010: ID 05dc:a205 Lexar Media, Inc. 
Bus 001 Device 007: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 006: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 005: ID 03eb:2404 Atmel Corp. 
Bus 001 Device 004: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Nothing strange it seems...

Anyway, if i try to remove everything inside the "MonitorHeatbed" function, it still doesn't work but it behaves in a different way, so i think that function is still involved in some way.

Using the last devel code it gets stuck at Send: M110, while removing the "MonitorHeatbed" function first of all it can't anymore autodecect the serial connection, and then , if i set it manually i see Send: M110 and then Recv: B004. It then disconnects after a while.

...Any ideas?

donovan6000, 2016-02-08:
It almost sounds like M3D Fio isn't being loaded by OctoPrint. Do you have an M3D Fio tab in OctoPrint's settings? Does OctoPrint's log show any errors? It's at ~/.octoprint/logs/octoprint.log. It'll probably show a lot of SerialExceptions, but those are kinda normal for M3D Fio.

It could be related to Python DBus. In M3D Fio V0.25 I assumed that all Python Linux distros have the DBus module installed. This caused issues #57 and #63, but the latest devel branch has [exception handling in place](https://github.com/donovan6000/M3D-Fio/blob/devel/octoprint_m3dfio/__init__.py#L82-L86) if DBus can't be imported. So make sure your running the version I just pushed a couple minutes ago.

You can also try removing [these lines](https://github.com/donovan6000/M3D-Fio/blob/devel/octoprint_m3dfio/__init__.py#L837-L839) that create the thread to monitor the heat bed. Although that thread shouldn't be using much of the CPU, it could potentially cause issues if being run on something like a Raspberry Pi.

DGdev91, 2016-02-08:
Ok, that was the problem.
It's because i used to install M3D-Fio directly using PIP instead of doing that from the Octoprint interface.
It worked anyway since that commit, so i tought it was the same.

You can close this, my fault ^^

closed 2016-02-08 by DGdev91