donovan6000/M33-Fio #68: Can't autodetect port after disconnecting

ra100, 2016-02-07:
After disconnecting via octorpint command, I cant connect with AUTO port detection.

error message:
`Unhandled firmware error
The firmware reported an unhandled error. Due to that OctoPrint disconnected. Error: Failed to autodetect serial port, please set it manually.`

log:
```
Changing monitoring state from 'Connecting' to 'Closed'
Connection closed, closing down monitor
Connecting to: /dev/ttyACM1
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6ebbc530, open=True>(port='/dev/ttyACM1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1397
Changing monitoring state from 'Connecting' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1397'
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Detecting serial port'
Failed to autodetect serial port
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port.'
Changing monitoring state from 'Error: Failed to autodetect serial port.' to 'Detecting serial port'
Serial port list: ['/dev/ttyUSB1', '/dev/ttyUSB0', '/dev/ttyACM0']
Connecting to: /dev/ttyUSB1
Error while connecting to /dev/ttyUSB1: 'Timeout'
Connecting to: /dev/ttyUSB0
Error while connecting to /dev/ttyUSB0: 'Timeout'
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Failed to autodetect serial port
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port.'
Changing monitoring state from 'Error: Failed to autodetect serial port.' to 'Detecting serial port'
Serial port list: ['/dev/ttyUSB1', '/dev/ttyUSB0', '/dev/ttyACM0']
Connecting to: /dev/ttyUSB1
Error while connecting to /dev/ttyUSB1: 'Timeout'
Connecting to: /dev/ttyUSB0
Error while connecting to /dev/ttyUSB0: 'Timeout'
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
```

donovan6000, 2016-02-08:
That error is actually being generated by the newest devel octoPrint, not by M3D Fio. However it does look like either the printer's USB failed to be enumerated by the Raspberry Pi or something caused M3D Fio to break since it's switching from using M3D Fio's autodetect function to the default one.

Does OctoPrint's log show anything unusual? ~/.octoprint/logs/octoprint.log. Can you access and communicate with the printer after this happens with screen? `screen /dev/ttyACM0 115200`

Do you ever see this message while that's happening? This is genertaed when M3D Fio's autodetect function fails to locate any potential ports with the correct VID and PID.
[image]

donovan6000, 2016-02-08:
Also does this happen every time you manually disconnect and try to reconnect?

ra100, 2016-02-08:
I'm running octoprint 1.2.9.
The message showed.
And I've now tried few times to connect and disconnect printer and it seems to work fine. So it seems it's not reproductible. But if it happens again, I'l try to connect via screen.

closed 2016-02-10 by ra100