donovan6000/M33-Fio #69: Octoprint says S3D print is outside of printable area

crazyquark, 2016-02-07:
Hello,

I just started using Simplify3D with octoprint on my M3D but I am having an issue iwth larger prints. If I slice the same part with Cura I get no error, bit with gcode from S3D I get an error about it being too big.
At first I thought it was because of the bri, but removing it did not help.
I think it's an issue with the positioning of the print.
I had to check the dangerous option in the settings and the gcode ran fine...
Which again, is weird.

donovan6000, 2016-02-08:
Can you upload the Gcode file that S3D generates so I can see why this is happening?

donovan6000, 2016-02-08:
This might be related to #40. So check to see if the intro/outro added to the G-code by S3D has coordinates that go out of bounds. [Here's](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L312-L334) M3D Fio's bounds to compare them to.

crazyquark, 2016-02-08:
The regular M3D spooler says the g-code is OK.
Here's the file: 
https://www.dropbox.com/s/zqy59p77nqgbz5w/c1%2B_bot_case.gcode?dl=1

crazyquark, 2016-02-08:
I think I know what the issue is - S3D shifts everything to the right for some reason. I am going to open a ticket with them now.

crazyquark, 2016-02-08:
OK, after attempting to compensate for the offset by shifting the object in S3D and using the GCode viewer in Octoprint I realized that Octoprint's idea of what the size of the build plate is also off.

donovan6000, 2016-02-08:
M3D Fio includes a centering model pre-processor that can also reposition a model if it still goes out of bounds after being centered, so being shifted to the right shouldn't matter.

These are the print's max/min dimensions before the centering model pre-processor
Max X low: 99.617
Max X med: 90.098
Max X hig: N/A
Max Y low: 114.467 (This is out of bounds since M3D Fio's allows a max Y of 105 in the low boundary tier)
Max Y med: 104.948
Max Y hig: N/A
Max Z: 13.607
Min X low: 20.381
Min X med: 0.0
Min X hig: N/A
Min Y low: 5.533
Min Y med: 15.052
Min Y hig: N/A
Min Z: 0.35

When I run the following, the extruder hits the back and grinds. So this wont print in its current orientation with my printer.
```
G28
G90
G0 Y114.467 F2000
```

And here's the maxs/mins after the centering model pre-processor
Max X low: 101.8085
Max X med: 92.2895
Max X hig: N/A
Max Y low: 105.0 (This is in bounds now)
Max Y med: 95.481
Max Y hig: N/A
Max Z: 13.607
Min X low: 22.5725
Min X med: 2.1915
Min X hig: N/A
Min Y low: -3.934 (However this is out of bounds now since M3D Fio allows a min Y of -2 in the low boundary tier)
Min Y med: 5.585
Min Y hig: N/A
Min Z: 0.35

And running the following moves the extruder almost past my printer's bed to the point that it would be partially printing on the frame.
```
G28
G90
G0 Y-3.934 F2000
```

-3.934 isn't very much less than -2, so you could probably get away with printing it like this.

You can also try positioning the model inside of M3D Fio's model editor so that you can physically see the boundaries, export the positioned model with the 'save' button, and then slice that in S3D.
[image]

donovan6000, 2016-02-08:
OctoPrint's G-code viewer uses the dimensions specified in the printer profile, and the M3D printer profile's dimensions are X108, Y107, and Z112. These are based off of the [boundaries that M3D Fio uses](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L312-L334), so it's the size of the printable area and not the size of the bed.

crazyquark, 2016-02-08:
I understand, I also contacted S3D about this.
I will attempt to save the model as you suggested and then slice it with S3D.

crazyquark, 2016-02-08:
OK, seems S3D's idea of the M3D's dimensions are wrong. I adjusted the bed to the dimensions you gave me and moved the brim closer and now it all fits.
Oh, and S3D's support never got back to me.

Open source has better support :).
I am going to close this, thank you for your kind help.

donovan6000, 2016-02-08:
Awesome! Glad you got it working.

They're probably using the dimensions from this image that M3D released, which aren't even the dimensions used by the M3D's official software. I just moved the extruder around and ran M114 commands to get its X and Y coordinates which allowed me to generate the accurate boundaries used in M3D Fio.

[image]

crazyquark, 2016-02-18:
Sorry to reopen this but I am still having problems on the latest M3DFio.
The Gcode viewer shows the model nicely centered and if I slice with the server's CuraEngine the model fits.
However, when imported over from S3D M3DFio says it cannot print it; the M3D spooler works.
Attaching the g-code+stl file.
Might be a height problem?
[voronoi_vase.zip](https://github.com/donovan6000/M3D-Fio/files/135767/voronoi_vase.zip)

donovan6000, 2016-02-18:
You forgot to include the g-code in the zip. It's just the stl twice.

crazyquark, 2016-02-18:
Sorry about that. I will re-attach it when I get home.

crazyquark, 2016-02-18:
[voronoi_vase.zip](https://github.com/donovan6000/M3D-Fio/files/136666/voronoi_vase.zip)

donovan6000, 2016-02-19:
It'll work if you enable M3D Fio's center model pre-processor. You'll still get the warning, "Object too large to center on print bed", which just means that the object had to be re-positioned after being centered to make sure that it stayed in the printable regions.

The source of this problem is line 364549 in the G-code file, G1 X0 F2160 ; move X axis to zero. This happens in the upper printable region, where the smallest X can be is 7, so you can also just remove that line and you wont have to use the center model pre-processor.

crazyquark, 2016-02-19:
Thank you, I will try that!

crazyquark, 2016-02-19:
I seem to have found a workaround: in S3D I reset it's own centering to 0,0,0 and then let M3DFio center the gcode for me. It's not great but it seems to work for now.

closed 2016-03-15 by donovan6000