donovan6000/M33-Fio #70: Disabling raft

totterfree, 2016-02-09:
This is more of a feature request than an issue, but I couldn't find somewhere more appropriate to put it. 
I'd like the ability to print without a raft. All of my prints so far have had a raft which is so well adhered to the model that it can't be removed without breaking the part. I'd like to be able to print without the raft so that I can have more flexibility to do parts which do not need them.

totterfree, 2016-02-09:
I've found a solution to my noobish problem. It seems that this is done by creating a profile which has no raft. The easiest way to do this is to open "C:\Users\USERNAME\AppData\Local\M3D\M3DSoftware\Working" and find default.cfg. This is the profile the M3D software creates whenever you click print. There is custom code at the bottom which must be removed in order for octoprint to parse it. It will look something like this... 

```
startCode = """
G90
;other temp:220
;ideal temp:220
"""
endCode = """
M18
"""
```

Remove that, and import via the Cura settngs in the octoprint UI.
All that said, it would be nice if M3D-Fio had a UI for creating these same profiles.

donovan6000, 2016-02-09:
The [devel branch](https://github.com/donovan6000/M3D-Fio/tree/devel) of M3D Fio implements a more advanced profile editor that lets you easily print without a raft, so the next release of M3D Fio will have what your asking for.
[image]

donovan6000, 2016-02-10:
Just released M3D Fio V0.26 and it has the new profile editor, so everything should be all good now :)

closed 2016-02-10 by donovan6000