donovan6000/M33-Fio #72: Wave bonding without raft/brim

ra100, 2016-02-10:
Is it possible to use wave bonding without raft? I know it's not standard, with wave bonding won't bottom layer look very nice, I think, but it would save a lot of material sometimes.
I tried to set wave+heat bonding with on-the-fly preprocessing, but when I print without raft, it prints without wavebonding.
This is maybe more of an feature request than bug.

donovan6000, 2016-02-10:
Based on the lighting on the model on the right that lets us slightly see its depth, these models don't look like they're using a rafts.
[image]
My wavebonding algorithm does differ from the official one. Back when M3D Linux and Microprint were just getting started, I asked tomasf about [the weird dot product function used in the official wavebonding](https://printm3d.com/community/viewtopic.php?f=5&t=1771&start=20#p19454), and he said that [Mike said it was a bug](https://printm3d.com/community/viewtopic.php?f=5&t=1771&start=20#p19469). So now 7 months later that bug hasn't been fixed, so I'm really starting to doubt what tomasf told me.

There's a few bugs I noticed in M3D Fio V0.26 that I'm fixing right now, so I'll also change the wavebonding to be more like the the official while I'm at it. That way we can see if that will make it work better when printing without a raft.

donovan6000, 2016-02-11:
Newest devel branch now has wave bonding that's more like the original

ra100, 2016-02-11:
Thank you, I'm gonna try it.

ra100, 2016-03-05:
Sorry, I forgot to report back. It happened maybe onece, twice, that wavebonding worked iwthout raft/brim, an maybe only on "low print quality". I mostly print on 0.35 layer and I think that it was always without wavebonding. And with brim it doesn't work all the time.
I'm using on-the-fly preprocessor with shared library.

Any ideas what could I try change in my settings to test wave bonding?

donovan6000, 2016-03-15:
I just compared the G-code when using wave bonding and when not using wave bonding, and it's definitely doing stuff. Wave bonding is the pre-processor that I understand the least about since I've never really looked into what it does.

M3D Fio's wave bonding still differs slightly from M3D's version so that might be the reason that it doesn't look like it's working all the time. M3D's wave and thermal bonding save the last command and compare it to the current command to determine if they make a sharp angle of movement. So M3D's software will use whatever the last command is, even if it's a comment or temperature command, in the sharp angle function, and that results in a lot of fake sharp angles. Only G0 and G1 commands can cause the extruder to move, so that's all that M3D Fio uses in the sharp angle function which results in only real sharp angles.

ra100, 2016-03-16:
Thanks for clarification. Don't know if anything changed, but now I think I can always see "up-down" movement on first layer with wave bonding.

closed 2016-03-16 by ra100