donovan6000/M33-Fio #74: Can't cold extrude message before print

ra100, 2016-02-11:
When trying to print something else than test border, I get warning about cold extruding. Than temperature starts to rise and head starts to move to BR corner at slow speed and won't stop (I didn't let it try to get there for long I was affraid of damaging printer), it's not like resetting position before print. Happend with on-the-fly and without, heat and wavebonding on.

octoprint.log
```
2016-02-11 19:25:26,550 - octoprint.plugins.cura - INFO - Running u'/usr/local/sbin/cura_engine -v -p -s coolHeadLift=1 -s downSkinCount=2 -s enableCombing=0 -s endCode=\n -s extrusionWidth=350.0 -s fanFullOnLayerNr=1 -s fanSpeedMax=255 -s fanSpeedMin=255 -s filamentDiameter=1750 -s filamentFlow=100 -s fixHorrible=0 -s gcodeFlavor=0 -s infillOverlap=0 -s infillSpeed=16 -s initialLayerSpeed=12 -s initialLayerThickness=300 -s initialSpeedupLayers=4 -s inset0Speed=12 -s insetCount=1 -s insetXSpeed=14 -s layer0extrusionWidth=350 -s layerThickness=350 -s minimalExtrusionBeforeRetraction=100 -s minimalFeedrate=12 -s minimalLayerTime=0 -s moveSpeed=20 -s multiVolumeOverlap=0 -s objectSink=0 -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added after the T(n)\n -s posx=52000 -s posy=58000 -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added before the T(n)\n -s printSpeed=16 -s retractionAmount=2400 -s retractionAmountExtruderSwitch=14500 -s retractionMinimalDistance=1500 -s retractionSpeed=20 -s retractionZHop=100 -s skirtDistance=2000 -s skirtLineCount=0 -s skirtMinLength=0 -s sparseInfillLineDistance=4000 -s startCode=M109 T0 S220.0\nT0\n; Sliced at: Thu 11-02-2016 19:25:26\n; Basic settings: Layer height: 0.35 Walls: 0.35 Fill: 8.75\n -s supportAngle=50 -s supportEverywhere=1 -s supportExtruder=-1 -s supportLineDistance=1750 -s supportType=1 -s supportXYDistance=700 -s supportZDistance=150 -s upSkinCount=2 -o /tmp/tmpYIcR8q.gco /media/nidhogg/pi/.octoprint/uploads/tienidlo/base1.stl' in /usr/local/sbin
```

serial.log
```
2016-02-11 19:25:47,011 - SERIAL - DEBUG - Changing monitoring state from 'Operational' to 'Printing'
2016-02-11 19:25:47,218 - SERIAL - DEBUG - Send: N0 M110 N0*125
2016-02-11 19:25:47,225 - SERIAL - DEBUG - Recv: ok 0
2016-02-11 19:25:47,227 - SERIAL - DEBUG - Send: N1 G1 E-2.40000 F1200*44
2016-02-11 19:25:47,241 - SERIAL - DEBUG - Recv: ok 1
2016-02-11 19:25:47,245 - SERIAL - DEBUG - Send: N2 M109 S245*105
2016-02-11 19:25:47,246 - SERIAL - DEBUG - Recv: ok Cannot cold extrude
2016-02-11 19:25:47,268 - SERIAL - DEBUG - Send: N3 G1 Z0.100000*78
2016-02-11 19:25:47,686 - SERIAL - DEBUG - Recv: T:142.1424
2016-02-11 19:25:48,733 - SERIAL - DEBUG - Recv: T:147.4669
2016-02-11 19:25:50,095 - SERIAL - DEBUG - Recv: T:152.8745
2016-02-11 19:25:50,728 - SERIAL - DEBUG - Recv: T:157.1837
2016-02-11 19:25:51,737 - SERIAL - DEBUG - Recv: T:157.1837
2016-02-11 19:25:52,714 - SERIAL - DEBUG - Recv: T:162.7543
2016-02-11 19:25:53,677 - SERIAL - DEBUG - Recv: T:162.7543
2016-02-11 19:25:54,659 - SERIAL - DEBUG - Recv: T:168.4141
2016-02-11 19:25:55,654 - SERIAL - DEBUG - Recv: T:172.9528
2016-02-11 19:25:56,650 - SERIAL - DEBUG - Recv: T:172.9528
2016-02-11 19:25:57,649 - SERIAL - DEBUG - Recv: T:172.9528
2016-02-11 19:25:58,648 - SERIAL - DEBUG - Recv: T:178.7875
2016-02-11 19:25:59,638 - SERIAL - DEBUG - Recv: T:178.7875
2016-02-11 19:26:00,635 - SERIAL - DEBUG - Recv: T:183.4857
2016-02-11 19:26:02,629 - SERIAL - DEBUG - Recv: T:189.5039
2016-02-11 19:26:03,624 - SERIAL - DEBUG - Recv: T:189.5039
2016-02-11 19:26:04,621 - SERIAL - DEBUG - Recv: T:189.5039
2016-02-11 19:26:05,615 - SERIAL - DEBUG - Recv: T:189.5039
2016-02-11 19:26:06,612 - SERIAL - DEBUG - Recv: T:195.6224
2016-02-11 19:26:07,608 - SERIAL - DEBUG - Recv: T:195.6224
2016-02-11 19:26:09,600 - SERIAL - DEBUG - Recv: T:200.5805
2016-02-11 19:26:10,596 - SERIAL - DEBUG - Recv: T:200.5805
2016-02-11 19:26:11,593 - SERIAL - DEBUG - Recv: T:205.6226
2016-02-11 19:26:12,589 - SERIAL - DEBUG - Recv: T:206.8966
2016-02-11 19:26:14,582 - SERIAL - DEBUG - Recv: T:212.0358
2016-02-11 19:26:15,577 - SERIAL - DEBUG - Recv: T:212.0358
2016-02-11 19:26:16,584 - SERIAL - DEBUG - Recv: T:212.0358
2016-02-11 19:26:17,569 - SERIAL - DEBUG - Recv: T:212.0358
2016-02-11 19:26:18,562 - SERIAL - DEBUG - Recv: T:217.2636
2016-02-11 19:26:19,562 - SERIAL - DEBUG - Recv: T:218.5595
2016-02-11 19:26:20,563 - SERIAL - DEBUG - Recv: T:218.5595
2016-02-11 19:26:21,551 - SERIAL - DEBUG - Recv: T:218.5595
2016-02-11 19:26:22,550 - SERIAL - DEBUG - Recv: T:223.8895
2016-02-11 19:26:23,566 - SERIAL - DEBUG - Recv: T:223.8895
2016-02-11 19:26:24,540 - SERIAL - DEBUG - Recv: T:223.8895
2016-02-11 19:26:25,536 - SERIAL - DEBUG - Recv: T:223.8895
2016-02-11 19:26:26,531 - SERIAL - DEBUG - Recv: T:223.8895
2016-02-11 19:26:27,532 - SERIAL - DEBUG - Recv: T:229.3131
2016-02-11 19:26:28,524 - SERIAL - DEBUG - Recv: T:229.3131
2016-02-11 19:26:29,522 - SERIAL - DEBUG - Recv: T:230.6315
2016-02-11 19:26:30,523 - SERIAL - DEBUG - Recv: T:230.6315
2016-02-11 19:26:31,515 - SERIAL - DEBUG - Recv: T:230.6315
2016-02-11 19:26:32,508 - SERIAL - DEBUG - Recv: T:230.6315
2016-02-11 19:26:33,507 - SERIAL - DEBUG - Recv: T:236.1627
2016-02-11 19:26:34,502 - SERIAL - DEBUG - Recv: T:236.1627
2016-02-11 19:26:35,499 - SERIAL - DEBUG - Recv: T:236.1627
2016-02-11 19:26:37,181 - SERIAL - DEBUG - Recv: ok 2
2016-02-11 19:26:37,196 - SERIAL - DEBUG - Send: N4 M105*35
2016-02-11 19:26:37,205 - SERIAL - DEBUG - Recv: ok 3
2016-02-11 19:26:37,216 - SERIAL - DEBUG - Send: N5 M104 S235*100
2016-02-11 19:26:37,221 - SERIAL - DEBUG - Recv: ok 4
2016-02-11 19:26:37,227 - SERIAL - DEBUG - Recv: T:243.1350
2016-02-11 19:26:37,242 - SERIAL - DEBUG - Send: N6 G0 X43.023000 Y49.978000 Z0.414306*67
[... 111 lines cut ...]
2016-02-11 19:27:08,394 - SERIAL - DEBUG - Recv: ok 2
2016-02-11 19:27:08,401 - SERIAL - DEBUG - Send: N1 G91
2016-02-11 19:27:08,411 - SERIAL - DEBUG - Recv: ok 0
2016-02-11 19:27:08,431 - SERIAL - DEBUG - Send: N2 G0 X-10 F3000
2016-02-11 19:27:08,474 - SERIAL - DEBUG - Recv: ok 1
2016-02-11 19:27:08,481 - SERIAL - DEBUG - Send: N0 M110
2016-02-11 19:27:08,492 - SERIAL - DEBUG - Recv: ok 2
2016-02-11 19:27:08,505 - SERIAL - DEBUG - Send: N1 G91
2016-02-11 19:27:08,520 - SERIAL - DEBUG - Recv: ok 0
2016-02-11 19:27:08,529 - SERIAL - DEBUG - Send: N2 G0 X-10 F3000
2016-02-11 19:27:08,564 - SERIAL - DEBUG - Recv: ok 1
2016-02-11 19:27:08,574 - SERIAL - DEBUG - Send: N0 M110
2016-02-11 19:27:08,589 - SERIAL - DEBUG - Recv: ok 2
2016-02-11 19:27:08,593 - SERIAL - DEBUG - Send: N1 G91
2016-02-11 19:27:08,621 - SERIAL - DEBUG - Recv: ok 0
2016-02-11 19:27:08,629 - SERIAL - DEBUG - Send: N2 G0 X-10 F3000
2016-02-11 19:27:08,676 - SERIAL - DEBUG - Recv: ok 1
2016-02-11 19:27:08,707 - SERIAL - DEBUG - Recv: ok 2
2016-02-11 19:27:11,456 - SERIAL - DEBUG - Recv: wait
2016-02-11 19:27:12,453 - SERIAL - DEBUG - Recv: 
2016-02-11 19:27:12,641 - SERIAL - DEBUG - Send: M105
2016-02-11 19:27:12,647 - SERIAL - DEBUG - Recv: ok
2016-02-11 19:27:12,651 - SERIAL - DEBUG - Recv: T:0
2016-02-11 19:27:13,446 - SERIAL - DEBUG - Recv: wait
```

donovan6000, 2016-02-12:
Is the preparation pre-processor enabled. When printing a test border or backlash cylinder, it forces it to use the preparation pre-processor, so that would explain why printing the test border doesn't show the 'cannot cold extrude error' but everything else does.

Also that serial output isn't using on-the-fly pre-processing. If it was, you'd see more double asterisks, **, before the command checksums since that's what M3D Fio uses to prevent pre-processing already processed commands. I know you said you tried it with and without it, so something isn't working right if that was the serial output when you had on-the-fly pre-processing enabled.

ra100, 2016-02-12:
Sorry, my bad, I wanted to turn off GPIO fan and I turned off whole Preparation.

closed 2016-02-12 by ra100