donovan6000/M33-Fio #79: Can't load, cv2 module missing

ra100, 2016-02-12:
I tried to install cv2 module (pip instal cv2), but that didn't help.

```
2016-02-12 19:00:19,169 - octoprint.plugin.core - ERROR - Error loading plugin m3dfio
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev888+geacd9f7-py2.7.egg/octoprint/plugin/core.py", line 650, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.26-py2.7.egg/octoprint_m3dfio/__init__.py", line 42, in <module>
    import cv2
ImportError: No module named cv2
```

donovan6000, 2016-02-12:
On my computer 'pip install cv2' didn't really do anything, so I had to do 'sudo apt-get install python-opencv' before I could import the cv2 module. I'll probably make it optional though and only implement the builtin webcam server if the user has the cv2 module installed since there's no reliable way to enforce having it installed and not everyone plans on using their webcam.

M3D Fio's opencv usage is pretty cool though. I was testing out OctoPrint's webcam support yesterday on Linux and Windows, and getting mjpg-streamer compiled wasn't easy since it's pretty old, and getting Yawcam installed required installing java too. So now M3D Fio can host the webcam on its own and automatically configure OctoPrint's settings to use the paths that it host for streaming and snapshots, so it's very easy to get OctoPrint's webcam supporting working now :)

donovan6000, 2016-02-12:
Ok, so the cv2 requirement is optional now. It'll just show/hide the camera settings depending on if it's installed or not.

ra100, 2016-02-13:
Nice feature and thanks for making it optional, I don't have webcam and don't need opencv on RPi for now.

closed 2016-02-13 by ra100