donovan6000/M33-Fio #80: Freeze on calibrating Z0 (Raspberry Pi/Octoprint)

jctj, 2016-02-14:
Using your excellent work, I have a Mac pushing to a Raspberry Pi 2 that controls the M3D via Octoprint. I have finally gotten things *almost* working, but whenever I calibrate via Octoprint, the website freezes (left it running for 30+ min) and I have to unplug everything. Had to move the head manually to change ink, and can't get Octoprint to see that the M3D is down and centered. 

I have gotten Octoprint to initiate the calibration and the head goes back to grind, right to grind, back to center, and then down to the bed with some up/down action going on, but then the M3D printer does nothing and the Octoprint website keeps the "calibration is process" warning up. 

And thank you so much for what you are doing - M3D is nearly worthless on a Mac and the Pi is a perfect server.

donovan6000, 2016-02-14:
What does OctoPrint's terminal tab show when this happens? You won't be able to go to that tab when the 'calibration in progress' message is showing, you'll need to refresh the webpage and then go to the controls tab.

jctj, 2016-02-15:
I fired everything back up again. Delays are unbelievable - as in it takes 10 min for Octopi.local/ to be able to successfully load (lots of failed attempts before it finally works, doing nothing but hitting the reload button all the while). Once it was ready, I checked terminal and it had: 
emperature
Control
GCode Viewer
Terminal
Timelapse
Send: M619 S41 T1
Recv: ok  PT:40 DT:67
Send: M619 S70 T4
Recv: ok  PT:41 DT:165
Send: M619 S74 T4
Recv: ok  PT:70 DT:0
Send: M619 S78 T4
Recv: ok  PT:74 DT:0
Send: M619 S82 T4
Recv: ok  PT:78 DT:0
Recv: ok  PT:82 DT:0
Send: M619 S86 T4
Send: M619 S94 T4
Recv: ok  PT:86 DT:0
Send: M619 S102 T4
Recv: ok  PT:94 DT:1153138688
Send: M619 S106 T4
Recv: ok  PT:102 DT:1153138688
Send: M619 S110 T4
Recv: ok  PT:106 DT:1153138688
Send: M619 S114 T4
Recv: ok  PT:110 DT:1114636288
Send: M619 S118 T4
Recv: ok  PT:114 DT:1120665600
Send: M619 S98 T1
Recv: ok  PT:118 DT:1135869952
Send: M420 T100
Recv: ok  PT:98 DT:1
Recv: ok
Recv: wait
Send: M105
Recv: ok
Recv: T:0
Recv: wait
Changing monitoring state from 'Operational' to 'Closed'
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x1fc0b30, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 250000
Send: M110
Recv: e1
Changing monitoring state from 'Detecting baudrate' to 'Closed'
Changing monitoring state from 'Offline' to 'Detecting serial port'
Connection closed, closing down monitor
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x2299270, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Recv: e1
Recv: wait
Trying baudrate: 250000
Send: M110
Recv: e1
Recv: e1
Recv: ok
Changing monitoring state from 'Detecting baudrate' to 'Operational'
Send: M21
Unexpected error while writing to serial port: SerialException: 'write failed: [Errno 5] Input/output error' @ comm.py:_doSendWithoutChecksum:1752
Connection closed, closing down monitor
[... 450 lines cut ...]
Send: M105
Recv: T:0.0000
Recv: ok
Recv: T:0
Recv: ok
Recv: T:0
Recv: ok
Recv: T:0
Recv: ok
Recv: T:0
Recv: wait
Send: N0 M110
Recv: ok 0
Send: M117
Recv: ok ZV:1 S:3
Send: N1 M65536
Send: M105
Recv: ok
Recv: T:0
Recv: wait ...
And deja vu all over again. Disconnected and reconnected, with nothing touching/changing via the hardware, cables, set up, etc.

Thanks for your help!

James

donovan6000, 2016-02-17:
It all looks normal. Does OctoPrint's log show any error? It's in ~/.octoprint/logs/octoprint.log

donovan6000, 2016-03-16:
I'm probably going to close this issue in the next couple of days since no ones responded to it in awhile. Let me know if you still have any interest in testing/solving it.

jctj, 2016-03-17:
Ugh - I *want* to but just bogged down so much at the moment. Please close it and when I get to fire the printer up again (with updated software, etc.) if there is still a problem I will let you know then. 

Thank you again for all the work you put into such a needed solution! Really, M3D should have done this (would be great if they paid you a gift or something…).

James

-- 
James Taylor
Sent with Airmail

donovan6000, 2016-03-17:
Ok

closed 2016-03-17 by donovan6000