donovan6000/M33-Fio #83: Can't connect to the printer

abbradar, 2016-02-17:
Hi,

I run OctoPrint 1.2.9 and M3D Fio 0.26. When trying to connect the printer, I get various failures -- I was able to get a successful connection once, but other times there are different (but bad) outcomes:

1.
```
Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM5
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x7f5ded67a210, open=True>(port='/dev/ttyACM5', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 250000
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
...
```

2.
```
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x7f5dec520f90, open=True>(port='/dev/ttyACM5', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Connection closed, closing down monitor
```

3.
```
Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM6
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x7f5ded697090, open=True>(port='/dev/ttyACM6', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Recv: wait
Recv: wait
...
Recv: wait
Recv: wait
Recv: e1
Recv: wait
```

Meanwhile, (M3D-Linux)[https://github.com/donovan6000/M3D-Linux] somewhat works -- I was able to update firmware and get serial number and firmware version of the device (although I haven't checked further and it wasn't able to get device settings).

donovan6000, 2016-02-17:
Try restarting the device and see if that fixes the problem. It kinda looks like M3D Fio might not have been successfully installed or isn't being loaded by OctoPrint. Do you have an M3D Fio tab in OctoPrint's settings?

Also make sure you have the [udev rule](https://github.com/donovan6000/M3D-Fio/blob/master/installers/Linux/90-m3d-local.rules) installed to make sure the correct read/write permission gets applied to the printer and to prevent programs like ModemManager from messing up communication.

abbradar, 2016-02-17:
Hm, it may be my own problems -- I'm making an OctoPrint package and service for NixOS (a Linux distribution) and I package M3D Fio customly, too. I do have the tab, and it looks that plugin is generally loaded (I have its profile enabled and so on). I don't see any exceptions in OctoPrint log (some were there earlier due to bad packaging). Is there some "verbose logging mode" in M3D Fio? Meanwhile I'll try adding random _print_ statements inside by myself and see what comes out.

abbradar, 2016-02-17:
BTW, I've finished the packaging: https://github.com/NixOS/nixpkgs/pull/13069. I also have a custom patch to build the C++ library from source (we don't like bundled binaries), place it to the standard `$PREFIX/lib` and access it from there instead of detecting the platform.

Thanks you very much for your work! The printer works flawlessly.

PR #84 (2016-02-17): Try to create connection several times if printer is not yet available
Fixes #83 for me. It gets "Permission denied" the first time -- seems like some race condition with udev or maybe the kernel returns EPERM when the device is not yet ready.
closed 2016-02-17 by donovan6000