donovan6000/M33-Fio #87: Disconnected from Server

neilferreri, 2016-02-18:
I get the following message more often than not:
"**You've been disconnected from the server. This page will now be refreshed to prevent any problems.**"
The only option is to click OK, and the page refreshes and we start all over.  This happens on multiple computers, phones, tablets, but it has also worked on all at some point.
It seems as though the connection is hanging when the error message pops up, as soon as I click OK, I see that it logs me in shows connection info & my uploaded files...then refresh to start the cycle over.

Any thoughts?

Thanks!﻿

donovan6000, 2016-02-18:
Does OctoPrint's log show any errors? It's in ~/.octoprint/logs/octoprint.log

neilferreri, 2016-02-19:
There is nothing logged when the error occurs.  I have not been able to
connect at all today due to this error.
Again, the raspberry Pi is connected via ethernet, and i can connect to the
Pi just fine.  Every time I open a browser to go connect to the printer, I
get the same message.
[image: Inline image 1]

This is the only plugin I've added.
I've cleared cache on all devices I've tried with no change.
I've rebooted the Pi.  It has been updated / upgraded.  I'm using the most
recent octopi image.

Thanks for any help!
Neil

donovan6000, 2016-02-19:
I have no idea. OctoPrint calls its onServerDisconnect hook whenever a client is disconnected from the server. [M3D Fio's function for that hook](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/static/js/m3dfio.js#L8111-L8155) displays the message that your seeing.

I've never used M3D Fio over a network or on a Raspberry Pi, so either of those might be contributing to this problem.

Freefall90, 2016-02-22:
I get the same issue. If (in Chrome) I manually clear the session ID and restart the browser, it usually will allow the page to load. I think this is due to the processing speed of the RPi, but I can't prove that yet. In some cases, I've had to reboot the Pi. Then I wait about 5 minutes to make sure everything is back up and running and reconnect to Octoprint.

neilferreri, 2016-02-22:
@donovan6000 Thanks for pointing me to the javascript file that displays that message.  I added a 2nd button to the message that just called hideMessage() without doing a reload (I may just remove the locationReload() call).  When I don't refresh, the connection seems fine.  I'm not sure, but it seems that when the network connection takes too long, that error come up.  I have no issues on my home network, but the setup that was giving me trouble was in a school.  I will just refresh on my own if I feel the need as it seemed unnecessary in this case.
If there's a reason why removing the reload is really bad, please let me know.  I am a hack...you clearly know what you are doing.
Thanks

donovan6000, 2016-02-22:
@neilferreri Nice! I might change it then to only recommend reloading or make it so that that message can't happen immediately after someone connects then.

The only times when not reloading would be bad is when your disconnected from the server while running an operation that is waiting on a response from the server, like loading filament, updating firmware, calibrating the bed, etc. In those cases, if you don't reload then the webpage would just hang forever.

neilferreri, 2016-02-22:
@donovan6000 I ended up just commenting out [this reload](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/static/js/m3dfio.js#L8467).  When I connect, I get the octoprint "Server Offline" error that says it will attempt a reconnect.  When it does, your message shows up, I dismiss it, and all is good.

donovan6000, 2016-02-25:
Ok, V0.27.1 has the choice of refreshing the page or closing the message.

closed 2016-02-25 by donovan6000